Arksine/katapult #6: Comand protocol should have a CRC?

KevinOConnor, 2022-05-04:
I may not fully understand the command protocol in CanBoot, so let me know if I got something wrong.  However, I didn't see a CRC check on commands and I'd be concerned about that.

It's possible for the canbus to reorder packets and/or drop some packets.  We've seen some host adapters do this in practice.  The fear is that a reordered data packet may subtly alter the flashed contents - the application (eg, Klipper) may actually even start and run with a subtle corruption like that, but is likely to cause hard to debug crashes.

Granted, the verification check should report a mismatch, but that check is also subject to reordering issues (that is it may spurious report failures even when the device is successfully flashed).  I've been testing locally with an old innomaker can adapter that has this reordering problem and it consistently fails the verification check (but seems to generally write out correct images).

I'm not sure I fully understand the current command protocol, but I wonder if we could add a CRC.  Maybe change the protocol to something like:
`<0x6362746C><1-byte length><1-byte command><data><16-bit crc><etx>`?  In this proposal both args and block data would be contained within the command message.  I'm proposing a 16-bit crc just because it's easy to grab that from the current Klipper code.  This would also fix a block size of 64 bytes.

I'd also suggest sending the flash write address with every flash block to avoid possible issues with dropped messages.  May also be useful for the ack message to state what command and address is being acknowledged.

A CRC check would also make it easier to support flashing over UART instead of CAN.  (Several users with stm32 and rp2040 boards have requested that feature and it shouldn't be too hard to extend the code to support Klipper's UART handlers instead of CAN.)

Thoughts?
-Kevin

Arksine, 2022-05-05:
The protocol is very simple and indeed the commands do not contain a CRC.   No command can exceed 8 bytes, so each fits within a single CAN frame.  Since each can frame already contains a CRC it seemed that adding one to the protocol was redundant.   The exception is the transmission of a block, which as you point out is subject to the verification check.

> It's possible for the canbus to reorder packets and/or drop some packets. We've seen some host adapters do this in practice. The fear is that a reordered data packet may subtly alter the flashed [...]

I was under the impression that this was a device issue rather than a canbus issue.  For example, the dropped frames we observe on the MCP2515 are the result the SPI device driver.    I do agree that adding a CRC to each block would detect these issues earlier, however I wonder if we can expect to reliably flash firmware on devices that exhibit this behavior (particularly that of the innomaker).

> I'd also suggest sending the flash write address with every flash block to avoid possible issues with dropped messages. May also be useful for the ack message to state what command and address is being acknowledged.

Agreed, this makes sense.

> A CRC check would also make it easier to support flashing over UART instead of CAN. (Several users with stm32 and rp2040 boards have requested that feature and it shouldn't be too hard to extend the code to support Klipper's UART handlers instead of CAN.)

If we want to support UART then I agree that CRC in the protocol becomes mandatory.

> <0x6362746C><1-byte length><1-byte command><data><16-bit crc><etx>? In this proposal both args and block data would be contained within the command message. I'm proposing a 16-bit crc just because it's easy to grab that from the current Klipper code. This would also fix a block size of 64 bytes (or at least, one not larger than about 250 bytes).

That is reasonable.  In this scenario I'm not sure if the etx is necessary since we have length?  I'll look into changing the protocol, I don't imagine it will be too difficult.

KevinOConnor, 2022-05-05:
Makes sense.

Separately, there are a few things I'm hoping to look at over the next few days:
1. Creating an automated "bootloader install application" - the ability to package a CanBoot bootloader installer into a binary that one can deploy with their existing bootloader.  (It's a pain to hook up stlink/serial to flash a bootloader.)
2. Add support for flashing over uart/usb instead of just canbus.  I have an stm32f042 board with usb that is really painful to flash today - having a small usb bootloader for the stm32f0 would be nice.
3. Perhaps support 4KiB bootloader size (might be useful on the stm32f0 line that has small flash).

If you're looking at the command protocol then I'll hold off on the uart/usb stuff.  (As that work would be dependent on the command protocol.)

Cheers,
-Kevin

KevinOConnor, 2022-05-05:
Also, I'm fine with merging the canboot changes to Klipper whenever you are ready.  If you think changes are coming then might be better to wait before merging into Klipper.  It's up to you.

-Kevin

Arksine, 2022-05-05:
Sounds good.  I do plan to make your suggested changes to the protocol, hopefully I'll have that done soon.  I agree that we should wait on merging in Klipper until it is complete.

KevinOConnor, 2022-05-05:
FWIW, it occurred to me that if the protocol was something like: `<0x01><0x88><1 byte word count><1 byte command id> <n number of 32bit words> <2-byte crc><0x99><0x03>` then it might make for easy parsing/encoding.  The command handler and transmit functions could just treat the message as an array of `uint32_t[]`.

Just thinking out loud.
-Kevin

KevinOConnor, 2022-05-09:
This should be resolved with the merge of #12 .  Thanks!

-Kevin

closed 2022-05-09 by KevinOConnor