Arksine/katapult #11: Verifying failed on GD32F103CB

Telekatz, 2022-05-06:
Hi,

I tested the bootloader with a GD32F103. Flashing a firmware worked, but an error occurred during verification. The firmware was flashed successfully anyway.

I found the problem in the canbus_send() function. Apparently the CODE[1:0] bits in the CAN_TSR register of the GD32F103 are not working as described in the datasheet. If a message longer than 8 bytes is sent and thus more than one transmit mailbox should be used at the same time, mailbox 0 was always given back as the next free mailbox via CODE[1:0], even if it was currently in use.

The problem can be solved by using the TME0, TME1 and TME2 bits to determine the next free mailbox:

```
// Transmit a packet
int
canbus_send(uint32_t id, uint32_t len, uint8_t *data)
{
    uint32_t tsr = SOC_CAN->TSR;
    if (!(tsr & (CAN_TSR_TME0|CAN_TSR_TME1|CAN_TSR_TME2))) {
        // No space in transmit fifo - enable tx irq
        irq_disable();
        SOC_CAN->IER |= CAN_IER_TMEIE;
        irq_enable();
        return -1;
    }
#if 0
    int mbox = (tsr & CAN_TSR_CODE) >> CAN_TSR_CODE_Pos;
#else
    int mbox;

    if((tsr & (CAN_TSR_TME0)))
      mbox = 0;
    else if((tsr & (CAN_TSR_TME1)))
      mbox = 1;
    else
      mbox = 2;

#endif

    CAN_TxMailBox_TypeDef *mb = &SOC_CAN->sTxMailBox[mbox];
```

Is it possible to add this bugfix for the GD32F103 and make it available via a Kconfig option?

Arksine, 2022-05-07:
Presumably this issue would affect Klipper as well?  I prefer to keep the platform specific code in sync with Klipper, deviating could cause issues down the line.

Telekatz, 2022-05-07:
Yes, this issue affect Klipper as well and can be fixed in the same way.

bondus, 2022-06-03:
GD must have failed/forgot to implement TSR_CODE.
The code that Talekatz posted to select an empty mailbox should work fine on an STM32 as well.

Arksine, 2022-06-03:
Thanks.  I made this change in 870200826561b150137913d42fd7edc6515229ff.  I also successfully tested on an stm32f103.  I will PR the change to Klipper soon.