Arksine/katapult #27: CanBoot specific USB id

KevinOConnor, 2022-06-03:
This is just a note that CanBoot is currently using Klipper's USB ids.  When I ported the USB code from Klipper, I just copied the ids (the Klipper id is 0x1d50:0x614 as defined in src/Kconfig).

It might be worthwhile to obtain a unique USB id dedicated to CanBoot.  I don't think it's a huge issue as Linux shows the "product name" in lsusb and /dev/serial/by-id/ , and I did change that to "CanBoot" as part of porting.

The process I used to get a unique USB id was to clone the https://github.com/openmoko/openmoko-usb-oui repo, add a line to usb_product_ids.psv , and then submit a PR with that change.  For reference: https://github.com/openmoko/openmoko-usb-oui/pull/11

I'd make the request for CanBoot, but I'm not sure if you need to be the author of the PR.

-Kevin

Arksine, 2022-06-04:
Thanks Kevin.  I made the request and it was granted very quickly.   CanBoot's ID is 0x6177, with the change made in b8c68933098db6ad020a84188ee56a8ad3d0dec7.

KevinOConnor, 2022-06-04:
Great!

-Kevin

closed 2022-06-04 by KevinOConnor