Arksine/katapult #30: Some SKR 1.4 Turbo observations

Sineos, 2022-06-22:
I flashed CanBoot on my SKR 1.4 Turbo and noted some items:

## Flashing Klipper

Building Klipper for the SKR 1.4 and having Smoothieware Bootloader enabled leads to:
```
ubuntu@ubuntu ~/klipper $ make flash FLASH_DEVICE=1d50:6177
  Flashing out/klipper.bin to 1d50:6177
sudo dfu-util -d ,1d50:6177 -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

Failed to flash to 1d50:6177: Error running dfu-util

If flashing Klipper to a Smoothieboard for the first time it may be
necessary to manually place the board into "bootloader mode" - press
and hold the "Play button" and then press and release the "Reset
button".

When a Smoothieboard is in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1d50:6015

Alternatively, one can flash a Smoothieboard via SD card - copy the
"out/klipper.bin" file to a file named "firmware.bin" on an SD card
and then restart the Smoothieboard with that SD card.

make: *** [src/lpc176x/Makefile:37: flash] Error 255
```

Flashing with `./lib/canboot/flash_can.py -d /dev/ttyACM0` works fine and board boots correctly

Building Klipper for the board but disabling the Smoothieware Bootloader leads to:

```
ubuntu@ubuntu ~/klipper $ make flash FLASH_DEVICE=/dev/ttyACM1
  Flashing out/klipper.bin to /dev/ttyACM1
Entering bootloader on /dev/ttyACM1
Device reconnect on /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0
/usr/bin/python3 lib/canboot/flash_can.py -d /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.3:1.0 -f out/klipper.bin

/home/ubuntu/klipper/lib/canboot/flash_can.py:589: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
/home/ubuntu/klipper/lib/canboot/flash_can.py:271: DeprecationWarning: There is no current event loop
  self._reader = asyncio.StreamReader(CAN_READER_LIMIT)
Attempting to connect to bootloader
CanBoot Connected
Protocol Version: 1.0.0
Block Size: 64 bytes
Application Start: 0x4000
MCU type: lpc1769
Flashing '/home/ubuntu/klipper/out/klipper.bin'...

[##################################################]

Write complete: 8 pages
Verifying (block count = 455)...

[##################################################]

Verification Complete: SHA = AB01A6F5AAF2371766A1CD467074D86B86DACD0D
CAN Flash Success
```

Flashing succeeds but the board does not boot.

## Flashing CanBoot with STLink and OpenOCD

```
ubuntu@ubuntu ~ $ sudo openocd -f interface/stlink.cfg -f board/mcb1700.cfg -c "program /home/ubuntu/CanBoot/out/canboot.bin verify exit"
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 10 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.254047
Info : lpc17xx.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for lpc17xx.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 10 kHz, using 5 kHz
Info : Unable to match requested speed 10 kHz, using 5 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x1fff0080 msp: 0x10001ffc
Info : Unable to match requested speed 12500 kHz, using 4000 kHz
Info : Unable to match requested speed 12500 kHz, using 4000 kHz
** Programming Started **
Warn : Boot verification checksum in image (0x00000109) to be written to flash is different from calculated vector checksum (0xefff7b4a).
Warn : OpenOCD will write the correct checksum. To remove this warning modify build tools on developer PC to inject correct LPC vector checksum.
** Programming Finished **
** Verify Started **
Error: checksum mismatch - attempting binary compare
** Verify Failed **
shutdown command invoked
```

I'm by no means an OpenOCD expert and the command may be complete rubbish:
* The flashing seems to succeed and CanBoot is usable
* The Warning about the Checksum and the failed verification makes me nervous

Arksine, 2022-06-22:
When using `make FLASH` with CanBoot it is necessary to specify the device location (ie: `/dev/ttyACM0`).  Currently specifying a device ID will always launch `dfu-util`.  This is something `flash_usb.py` could probably handle better, Kevin briefly discussed some modifications that are related in [this pull request](https://github.com/Klipper3d/klipper/pull/5575).

Right now CanBoot uses the bootloader offset that Smoothieware uses, so the smoothieware bootloader needs to be checked in the Klipper build.  CanBoot doesn't need the full 16KiB Smoothie does, so we might want to look into adding an additional bootloader offset.

With regard to the checksum verification, its a detail specific to lpc176x processors.  The programming utility is responsible for calculating and writing the checksum during the upload process.  The `lpc21isp` and `flashmagic` utilities handle this correctly when programming over UART, it appears that OpenOCD does not.

Sineos, 2022-06-23:
Thanks for the insights. Appreciated.
Flashing via the device ID would generally be more comfortable as it can easily be retrieved via `lsusb`

closed 2022-06-23 by Sineos