Arksine/katapult #31: SKR V1.4 Turbo unable to reflash klipper

Bazeone, 2022-06-23:
I was able to flash Canboot onto the SKR V1.4 using a raspberry pi connected to the board via UART using the tft pins and lpc21isp command below.
lpc21isp -wipe -bin ~/CanBoot/out/canboot.bin /dev/ttyAMA0 230400 12000
After I was able to flash klipper with the following command.
python3 ~/CanBoot/scripts/flash_can.py -d /dev/ttyAMA0 -b 250000 -v -f ~/klipper/out/klipper.bin

After the first flash if I want to reflash or flash an updated version I have to double tap the reset button. Is this the normal Canboot behavior? I was hoping to be able to flash klipper via the command line without having to access the board itself.

I get the following error if I don't double tap the reset button.

Attempting to connect to bootloader
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError

Thank you for any information you may provide.

Arksine, 2022-06-23:
You must use `make FLASH FLASH_DEVICE=/dev/<device_name>`, where `<device_name>` is the specific device (ie: `ttyACM0`).  The `flash_can` script can't request the USB bootloader, so you can't run it directly unless CanBoot is running.

Bazeone, 2022-06-23:
﻿﻿make flash FLASH_DEVICE=/dev/ttyAMA0
gives me the following error.

Flashing out/klipper.bin to /dev/ttyAMA0

Failed to flash to /dev/ttyAMA0: Unable to find serial 'by-path' folder

If flashing Klipper to a Smoothieboard for the first time it may be
necessary to manually place the board into "bootloader mode" - press
and hold the "Play button" and then press and release the "Reset
button".

When a Smoothieboard is in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1d50:6015

Alternatively, one can flash a Smoothieboard via SD card - copy the
"out/klipper.bin" file to a file named "firmware.bin" on an SD card
and then restart the Smoothieboard with that SD card.

make: *** [src/lpc176x/Makefile:37: flash] Error 255

Thank you for your reply

Arksine, 2022-06-23:
Ah, I may have misread your OP.  I got that you uploaded CanBoot using the Pi's UART, however it appears that you have enabled UART communication for CanBoot and Klipper as well.  If that is the case, then at the moment there is indeed no way to request the bootloader over UART, this is going to require some changes to Klipper.  So indeed you will have to manually enter the bootloader using the "double click" method.

You may be better off using the Smoothieware bootloader and `flash-sdcard.py` until Kevin and I can come up with a method to do this.

Bazeone, 2022-06-23:
Thank you for your help, I didn't know about flash-sdcard.py. I was able to reflash klipper using flash-sdcard.py without having to reinstall the smoothieware bootloader.

Polar-Ted, 2022-11-11:
> You must use `make FLASH FLASH_DEVICE=/dev/<device_name>`, where `<device_name>` is the specific device (ie: `ttyACM0`). The `flash_can` script can't request the USB bootloader, so you can't run it directly unless CanBoot is running.

I assume this still true today? I've been testing Canboot on RP2040 and can't get back into the bootloader after flashing klipper on USB unless I double tap the reset or plug/replug the usb quickly.

Arksine, 2022-11-11:
> I assume this still true today? I've been testing Canboot on RP2040 and can't get back into the bootloader after flashing klipper on USB unless I double tap the reset or plug/replug the usb quickly.

I don't believe it will work with the RP2040.  Klipper's `flash_usb.py` script will need to be modified.

Polar-Ted, 2022-11-11:
> I don't believe it will work with the RP2040. Klipper's `flash_usb.py` script will need to be modified.

On a whim I tried using the klipper flash_usb.py. That script will fail to flash but it does manage to kick the RP2040 into the CanBoot bootloader. 

```
pi@vzero:~/klipper $ ls /dev/serial/by-id
usb-Klipper_rp2040_45503571279020E8-if00
pi@vzero:~/klipper $ python3 ~/klipper/scripts/flash_usb.py -t rp2040 -d /dev/serial/by-id/usb-Klipper_rp2040_45503571279020E8-if00  ~/klipper/out/klipper.bin
Entering bootloader on /dev/serial/by-id/usb-Klipper_rp2040_45503571279020E8-if00
Device reconnect on /sys/devices/platform/soc/3f980000.usb/usb1/1-1
sudo lib/rp2040_flash/rp2040_flash /home/pi/klipper/out/klipper.bin 1 11

Loaded UF2 image with 0 pages
No rp2040 in BOOTSEL mode was found
pi@vzero:~/klipper $ ls /dev/serial/by-id                                       usb-CanBoot_rp2040_45503571279020E8-if00
pi@vzero:~/klipper $
```

Arksine, 2022-11-11:
Right, `flash_usb.py` can request the bootloader, however the script currently assumes that pico_flash should be used to upload Klipper.  The build also needs to be modified to provide the .bin rather that a  .uf2 when canboot is installed.