Arksine/katapult #32: Huvud no longer updates via flash_can.py

JamesH1978, 2022-06-24:
Just rebuilt my system, can connection to huvud works well, but when i try and flash the newly made firmware to the huvud it doesn't seem to enter the bootloader and times out. This has worked in the past. Any insights? This is not all the output of the error, if you want it all, i will add it. I am hoping this is enough and its straightforward fix or a "i've been a muppet" ,

After this command, huvud is unresponsive until a power cycle

Thanks
James

pi@printpii:~/CanBoot/scripts $ service klipper stop
pi@printpii:~/CanBoot/scripts $ python3 flash_can.py -i can0 -f ~/klipper/out/klipper.bin -u 46abbf984311                                                                                                                                                        
Sending bootloader jump command...
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 46abbf984311, Application: CanBoot
Attempting to connect to bootloader
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError`

NAPCAL, 2022-06-24:
The stock HUVUD has HID bootloader on it; I haven't been able to get it into a true DFU mode to use STMCubeProgramer to wipe the flash memory and install CanBoot. I think I have been told you must have an ST-Link to force DFU to get CanBoot on the HUVUD.

JamesH1978, 2022-06-24:
I already have CANBoot on it, i flashed it with the programmer, the line returns:

Checking for canboot nodes...
Detected UUID: 46abbf984311, Application: CanBoot

So i know it has CanBoot on it, but it wont enter the bootloader and flash the new klipper version

NAPCAL, 2022-06-24:
What are you using for the CAN bus interface for your RPi?

JamesH1978, 2022-06-24:
waveshare can/rs485 hat

NAPCAL, 2022-06-24:
8Mhz or 12Mhz crystal?

JamesH1978, 2022-06-24:
12mhz, 8mhz was the old one

NAPCAL, 2022-06-24:
I have the 12Mhz Waveshare and BIQU Hermit Crab CAN Hat, currently using the HC CAN hat. 

I would like to see if I can duplicate your setup. How did you get CanBoot to replace the HID on the HUVUD? 
I have the 0.61 version.

NAPCAL, 2022-06-24:
Also, did you install a 120ohm resistor at the HUVUD connector on the CAN L & H?

JamesH1978, 2022-06-24:
i used a st-link v2 programmer to flash the bootloader to the huvud, which is a 0.61 too. and yes i soldered a 120r SMD resistor to the huvud, and made sure there was one present on the waveshare, proper termination is very important in can networks.

Arksine, 2022-06-24:
My guess is that you have an older version of CanBoot on your HUVUD that isn't compatible with the latest `flash_can.py` script.  It is definitely in the bootloader since CanBoot identifies itself.  However Kevin and I made some protocol changes, and it seems like the `connect` command isn't returned the expected result.

JamesH1978, 2022-06-24:
right, ok, so i should plug in the st link and re flash canboot then. I flashed it not long before you merged it into klipper

NAPCAL, 2022-06-24:
Just ordered a ST-Link V2, I will get it on Monday. I will see where I can get and if I can replicate your issue.

JamesH1978, 2022-06-25:
Just updated CanBoot on the huvud, and looks to now take the klipper.bin, but there seems to be some error output, but as i said it also succeeds... I assume this is note no.2 in action, as i use a 12mhz waveshare

Thanks
James

pi@printpii:~/CanBoot/scripts $ python3 flash_can.py -i can0 -f ~/klipper/out/klipper.bin -u 46abbf984311
Sending bootloader jump command...
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 46abbf984311, Application: CanBoot
Attempting to connect to bootloader
CanBoot Connected
Protocol Version: 1.0.0
Block Size: 64 bytes
Application Start: 0x8002000
MCU type: stm32f103xe
Verifying canbus connection
Flashing '/home/pi/klipper/out/klipper.bin'...

[##################################################]

Write complete: 25 pages
Verifying (block count = 399)...

[#######ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
###############ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
#################ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
###########]

Verification Complete: SHA = AE41990FD55717092214BF7711670062A6E5341D
CAN Flash Success

Arksine, 2022-06-25:
Yeah, this is just the waveshare canhat dropping frames during the verification process.  The script re-requests blocks that error out.  We could probably do a bit better with the output here so the tracebacks are only printed if the entire process fails.

JamesH1978, 2022-06-25:
thanks, can close this now :)

closed 2022-06-25 by JamesH1978