Arksine/katapult #34: [Feature Request] add an user definable device name

NAPCAL, 2022-07-01:
Add a user-definable device name that will be provided with the CAN bus ID when running flash_can.py -q
Since the direction for using the CAN bus will head toward multiple print heads, this will help debug and update these devices. 

Example output
pi@Fluidd:~ $ python3 /home/pi/CanBoot/scripts/flash_can.py -q
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 93499ef89c38, Application: Klipper, Device: Octopus Pro
Detected UUID: a0c90a3740c9, Application: Klipper, Device: Hermit Crab 1
Detected UUID: a0c90a3740d3, Application: Klipper, Device: Hermit Crab 2
Detected UUID: a0c90a3740f9, Application: Klipper, Device: Hermit Crab 3
Query Complete

krautech, 2024-11-28:
I recently looked into this myself. It seems klipper itself handles the naming of the device like that. So it'd be something that needs integration on both sides.

NAPCAL, 2024-11-28:
Klipper doesn't name the device.
I am talking about when the MCU responds to the Katapult query command (-q).