Arksine/katapult #35: Support for STM32H743 on the BTT SKR 3 and SKR 3 EZ

NAPCAL, 2022-07-10:
Please provide support for STM32H743; this is a processor being used on the NEW BigTreeTech boards.
CanBoot makes it easy to update without jumping through hoops or SD cards.
I am using it on all of my BigTreeTech boards with STM32 processors.

idontknowtom, 2022-08-29:
@Arksine @KevinOConnor With STM32H743 now supported by Klipper for FDCAN, is CanBoot support a possibility?

As I understand, CanBoot currently doesn't support FDCAN. For instance the STM32G0B1 gets configured to run standard CAN. Was hoping the same can be implemented with the H743 chip. Can provided any support as I currently have a SKR3 EZ with this processor...connected to an EBB36 (STM32G0B1) with Klipper USB to CAN bus bridge.

NAPCAL, 2022-08-29:
Please add standard CAN bus support for this ARM processor as was done for the STM32G0B1.

Klipper has provided standard CAN bus for STM32H743; it would be greatly appreciated if CanBoot could also be added to the SKR3/3EZ. Thanks in advance.

pnewb, 2022-10-03:
@Arksine would it help to send you an SKR3 and a prepaid return label?

idontknowtom, 2022-10-03:
@pnewb @Arksine Id be happy to send a SKR3 with return label as well but being based in Australia I’m not sure it would financially make any sense.

I could help anyone wanting to do this though.

acosano, 2022-11-07:
> Its possible, however BTT or someone that has a board with the STM32H743 would need to submit the code that implements in application programming. It may be similar to that of one of the supported models, however it would be impossible to test and debug without having the board on hand.

Hi m8. Where are you located and can you use a Fly Super 8 with a STM32F407ZGT6 chip to do this. Also do oyu need stepper drivers or can you use yours?

NAPCAL, 2022-11-07:
@Arksine or @KevinOConnor If one of you doesn't have an STM32H743-based board, please let me know. I have an extra BTT SKR3 that I can ship to one of you; just let me know.

NAPCAL, 2022-11-16:
> @Arksine or @KevinOConnor If one of you doesn't have an STM32H743-based board, please let me know. I have an extra BTT SKR3 that I can ship to one of you; just let me know.

I now have a 2nd spare BTT SKR3 from a BigTreeTech giveaway; I can offer it free of charge to the cause of the development of CanBoot & Klipper for STM32H743 support.

Eloura74, 2022-11-27:
Hi, I have a mini E3V3 motherboard with an rpi4, as well as the SB 2040 and a Can utoc3 3....
I once managed to get the Can with the lsusb command, but the flash didn't work, I then tried again but since my SB2040 is no longer listed, even in USB on my pc .... can you help me?

NAPCAL, 2022-11-27:
@Eloura74 not the correct thread for your issue; please find the RP2040 thread and post it there; you will get more responses.

KevinOConnor, 2022-12-16:
If interested in testing, see PR https://github.com/Arksine/CanBoot/pull/57.

-Kevin

NAPCAL, 2023-06-30:
Working as of June 29, 2023

closed 2023-06-30 by NAPCAL