Arksine/katapult #40: CanBoot Moonraker Update Manager

JonathanHeinz, 2022-08-20:
Hello,

maybe an update manager entry would be a cool feature, but I'm not sure if it would be such a useful feature, CanBoot updates should happen very rarely.

Arksine, 2022-08-25:
As you mentioned, it probably isn't something that needs updating often.  However it should be fairly simple to add an entry:

```
[update_manager CanBoot]
type: git_repo
origin: https://github.com/Arksine/CanBoot.git
path: ~/CanBoot
```
This assumes that you have the repo cloned at `~/CanBoot`.  One thing to keep in mind is that paths are case sensitive.

idontknowtom, 2022-08-26:
The only line I would add is the "is_system_service" line below. Testing this out and CanBoot was updating properly, but was giving an error due to trying to restart a service that doesn't exist.

```
[update_manager CanBoot]
type: git_repo
origin: https://github.com/Arksine/CanBoot.git
path: ~/CanBoot
is_system_service: False
```

NAPCAL, 2022-08-26:
Cool, thanks for the information.

JonathanHeinz, 2022-08-26:
> The only line I would add is the "is_system_service" line below. Testing this out and CanBoot was updating properly, but was giving an error due to trying to restart a service that doesn't exist. [...]

Thats why i opened a ticket, i didn't know that there is such an option, thank you.

NAPCAL, 2022-08-26:
The Moonraker information and documentation can be found here.

https://github.com/Arksine/moonraker

midmadn, 2023-10-09:
How does this work with katapult?

Arksine, 2023-10-09:
Presuming a new clone of Katapult in the home directory the following should work:

```
[update_manager katapult]
type: git_repo
origin: https://github.com/Arksine/katapult.git
path: ~/katapult
is_system_service: False
```