Arksine/katapult #42: CanBoot for RP2040

JonathanHeinz, 2022-08-29:
Possibility to install CanBoot on a RP2040 Chip

bwshockley, 2022-09-08:
I will second this request, with the understanding this is a big request.  I'm specifically using the Mellow Fly-SB2040 board and suspect other CANBUS boards will be moving to the RP2040.

tillin9, 2022-10-06:
I'll also +1 this request as I have an SB2040. Thanks all.

sh83, 2022-10-20:
This is my attempt on this: 
https://github.com/Arksine/CanBoot/pull/45

akhamar, 2022-10-26:
Would be nice indeed !

JonathanHeinz, 2022-11-07:
> This is my attempt: #45

Can i close the ticket now?

sh83, 2022-11-08:
> Can i close the ticket now?
I think it should be closed when klipper pr ( https://github.com/Klipper3d/klipper/pull/5874 ) will be merged.

jtrmal, 2022-11-08:
> I think it should be closed when klipper pr ( Klipper3d/klipper#5874

been merged just a while ago.
y.

closed 2022-11-08 by JonathanHeinz