Arksine/katapult #43: EBB42 CAN V1.1 Klipper Won't connect

ArketypeDesign, 2022-09-05:
I have followed the guide (and Eddies videos) to install canboot on the EBB42 CAN V1.1.
I used the menuconfig options from the BTT documentation.
I happen to be using the waveshare can hat.
After flashing canboot on the board, I successfully flashed klipper over the can bus.

pi@arkwire2:~/klipper/lib/canboot_ $ python3 flash_can.py -i can0 -f ~/klipper/out/klipper.bin -u 3266f8206791
Sending bootloader jump command...
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 3266f8206791, Application: CanBoot
Attempting to connect to bootloader
CanBoot Connected
Protocol Version: 1.0.0
Block Size: 64 bytes
Application Start: 0x8002000
MCU type: stm32g0b1xx
Verifying canbus connection
Flashing '/home/pi/klipper/out/klipper.bin'...

[##################################################]

Write complete: 13 pages
Verifying (block count = 411)...

[##################################################]

Verification Complete: SHA = 68F2BECD8F5259EC79195152106770D87BCD6D4D
CAN Flash Success

But upon restarting, the printer does not see the MCU EBBCan.

I have used a stripped down printer.cfg for testing.
Klippy log listed below.

Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'v0.10.0-566-g9f8f3cd7'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '3.7.3 (default, Jan 22 2021, 20:04:44) \n[GCC 8.3.0]'
Start printer at Mon Sep  5 09:15:58 2022 (1662383758.8 20.0)
===== Config file =====
[mcu]
serial = /dev/serial/by-id/usb-Klipper_stm32f103xe_39FFD7054247333439710757-if00
restart_method = command

[mcu EBBCan]
canbus_uuid = 3266f8206791

[printer]
kinematics = none
max_velocity = 100
max_accel = 100
=======================
mcu 'mcu': Starting serial connect
Loaded MCU 'mcu' 105 commands (v0.10.0-566-g9f8f3cd7 / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)
MCU 'mcu' config: ADC_MAX=4095 BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c1a=PB8,PB9 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_spi1=PA6,PA7,PA5 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi3=PB4,PB5,PB3 CLOCK_FREQ=72000000 INITIAL_PINS=!PC13 MCU=stm32f103xe PWM_MAX=255 RESERVE_PINS_USB=PA11,PA12 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
mcu 'EBBCan': Starting CAN connect
Created a socket
webhooks client 3045371664: New connection
webhooks client 3045371664: Client info {'program': 'Moonraker', 'version': 'v0.7.1-659-gf047167'}
mcu 'EBBCan': Timeout on connect
Created a socket
mcu 'EBBCan': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 259, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 317, in get_response
    cmd_queue)
[... 213 lines cut ...]
CONFIG_STM32_CANBUS_PB0_PB1=y
CONFIG_USB_VENDOR_ID=0x1d50
CONFIG_USB_DEVICE_ID=0x614e
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_CANSERIAL=y
CONFIG_CANBUS=y
CONFIG_CANBUS_FREQUENCY=250000
CONFIG_CANBUS_FILTER=y
CONFIG_INITIAL_PINS=""
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
CONFIG_HAVE_GPIO_HARD_PWM=y
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_HAVE_STRICT_TIMING=y
CONFIG_HAVE_CHIPID=y
CONFIG_HAVE_STEPPER_BOTH_EDGE=y
CONFIG_INLINE_STEPPER_HACK=y
=======================
Build file /home/pi/klipper/klippy/../out/klipper.dict(7674): Mon Sep  5 08:27:52 2022
Last MCU build version: v0.10.0-566-g9f8f3cd7
Last MCU build tools: gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1
Last MCU build config: ADC_MAX=4095 BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c1a=PF1,PF0 BUS_PINS_spi1=PA6,PA7,PA5 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi2a=PC2,PC3,PB10 BUS_PINS_spi3=PB4,PB5,PB3 CANBUS_FREQUENCY=250000 CLOCK_FREQ=64000000 MCU=stm32g0b1xx PWM_MAX=255 RECEIVE_WINDOW=192 RESERVE_PINS_CAN=PB0,PB1 RESERVE_PINS_crystal=PF0,PF1 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
Build file /home/pi/klipper/klippy/../out/klipper.elf(373984): Mon Sep  5 08:27:57 2022

NAPCAL, 2022-09-05:
You should be flashing from ~/CanBoot/scripts

Just an FYI flash_can.py will default to can0 and klipper.bin from ~/klipper/out
cd ~/CanBoot/scripts
python3 flash_can.py -u 3266f8206791

make sure you have the same CAN bus speed for all nodes including CanBoot. The make menuconfig will default to 500000 unless you have changed this setting during one of your make menuconfig. Also if the wrong STM clock is selected then the firmware will not respond.

make sure you selected all of the same settings for CanBoot and Klipper builds.

Arksine, 2022-09-05:
Check the bootloader offset for your Klipper build.   Looks like you configured CanBoot with an 8KiB application start, so Klipper's bootloader offset needs to match that.

ArketypeDesign, 2022-09-05:
I think it was the bootloader offset not set correctly... 
I quadruple checked those setings, and somehow still got it wrong.
 it looks like that worked Thanks!

closed 2022-09-05 by ArketypeDesign