Arksine/katapult #58: Firmware "Disapering" on CAN Board with RP2040 Chip after Reboot

LeergutLarss, 2022-12-21:
Hello

I instaled CanBoot + Klipper on my SB2040 Toolboard 
Now every time I restart everything the Toolboard is not recognized.
But for some reason if I connect my Toolboard after the Pi boots to USB and power everything works just fine.

MarvinBeym, 2023-01-03:
This may not be an issue caused by CanBoot as it happens on my 
"ercf eazy board CAN" (based on the rp2040 chip) (it's using the CAN interface) made by mellow (recently released) 

and it happens both with canboot + klipper as well as just klipper on its own (no bootloader offset)
(My board seams to have a possibly separate issue with can communication in general (no uuid at all))

Can confirm the first part, haven't tested what happens when board is connected after rpi is booted. Will try that tomorrow too.

However. Everytime the rpi is rebooted the firmware appears to be "gone" in the board in some way.

It will no longer show up in ls ....dev/by-id/....
I verified it further by defining a pin as the "gpio pin to turn on when booting (make menuconfig)".
When the rpi reboots, the led turns off and will never turn back on.

sh83, 2023-01-14:
Do anyone have reproducible case with canboot problem on rp2040 with up-to-date canboot/klipper? And with standalone klipper itself run fine on same hardware ( klipper can be used without canboot if it configured so).

Probably it was some problems, but I do not remember the way to reproduce them.

jtrmal, 2023-01-15:
I think it was always the other way around. Canboot was working but
standalone klipper wasn't.
Y.

hvraven, 2023-05-21:
I have this problem on a Mellow Fly SB2040 V1 using a RP2040 chip. I flashed the latest CanBoot, can flash the latest Klipper and it then works nicely. Until I power cycle the system, then I have to re-flash klipper using the still existing canboot.

I gladly test things if I am told what to do, currently I have no clue where to start looking into this problem.

MarvinBeym, 2023-05-21:
(should) be fixed by now:

https://github.com/Klipper3d/klipper/commit/bec1d957d216986da8da216aca557c4d31623fd2