Arksine/katapult #59: How to delect canboot from octoboard

Ct87, 2022-12-22:
Hello

i tried to set up ebb36 but I fail. I just saw it one time on my mainsail. After reboot it did not work .so I want to go without canbus. How do I flash back the octoboard  to stok settings ?

NAPCAL, 2022-12-22:
Unless you dumped the stock bootloader before putting CanBoot in place then you will need someone to dump theirs from the same matching STM32.

They will need to know if yours is Octopus V1 or Pro and the STM32 version F429, F446, H723, F407

Ct87, 2022-12-22:
Would it be possible to set klipper via canboot without canbus ... like setting up a new printer and then flash via uuid?
or would this make a bigger error?

NAPCAL, 2022-12-22:
You can setup Klipper with No bootloader, would need DFU to load it, I would use STM32CubeProg and wipe the flash before loading Klipper. Note: this would be the same process to update Klipper each time you would need to do so.

idontknowtom, 2022-12-22:
Bigtreetech doesn’t usually share the bootloader for their boards, but do share all Octopus bootloaders. They even have the new H723 chip bootloader.

Follow the PDF instructions you need STM32 Cube Programmer, USB-C cable, and some jumpers. Use the F407 file.

https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/tree/master/Firmware/DFU%20Update%20bootloader

Creo2005, 2022-12-22:
> You can setup Klipper with No bootloader, would need DFU to load it, I would use STM32CubeProg and wipe the flash before loading Klipper. Note: this would be the same process to update Klipper each time you would need to do so.

It is possible too flash MKS Robin Nano 1.2 with canbootloader? 
So that I can then flash the Klipper over the network.

idontknowtom, 2022-12-22:
> > You can setup Klipper with No bootloader, would need DFU to load it, I would use STM32CubeProg and wipe the flash before loading Klipper. Note: this would be the same process to update Klipper each time you would need to do so.
> 
> It is possible too flash MKS Robin Nano 1.2 with canbootloader? So that I can then flash the Klipper over the network. And how to do it right?

Yes the STM32F103 chip on your board is supported

Creo2005, 2022-12-22:
> Yes the STM32F103 chip on your board is supported

And how to do it right?

NAPCAL, 2022-12-22:
Thanks for the info on the Octopus bootloaders.

closed 2022-12-22 by Ct87