Arksine/katapult #61: BTT Manta M8P v1.1 CanBoot problem STM32G0B1

eibo68, 2023-01-02:
Having trouble getting CanBoot and klipper run on Manta M8P STM32G0B1 with CM4. Reset of the board gives me can0 interface, "~/CanBoot/scripts/flash_can.py -i can0 -q" gives me the uuid (if I'm quick enough) and then it disappears and not coming back before next reset of MCU. klipper service is shutdown while testing. Update of klipper with "~/CanBoot/scripts/flash_can.py -i can0 -u UUID" fails.
To me it looks like CAN on klipper is not working correct on the STM32G0B1 MCU. This is both with CAN-Hat and with "USB to CAN bus bridge (USB on PA11/PA12)"

bddanford, 2023-01-26:
You make any progress on this? I am pretty much having the same issues. Once I flash Klipper with the CAN bridge firmware on the MCU it disapears and I dont see it from the CM4 anymore. I don't know if I need too, or how to install CanBoot on the MCU. Do I need it? Will it show as a Can MCU or a normal MCU after the klipper CAN bridge firmware installed.

eibo68, 2023-01-26:
@bddanford Yeah, I have it up and running now. If you have the Manta M8P v1.1 with CANbus onboard:

cd ~/CanBoot/

    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32G0B1)  --->
    Build CanBoot deployment application (Do not build)  --->
    Clock Reference (8 MHz crystal)  --->
    Communication interface (CAN bus (on PD12/PD13))  --->
    Application start offset (8KiB offset)  --->
(250000) CAN bus speed
()  GPIO pins to set on bootloader entry
[*] Support bootloader entry on rapid double click of reset button
[ ] Enable bootloader entry on button (or gpio) state
[ ] Enable Status LED

make clean
make

cd klipper
make menuconfig

[*] Enable extra low-level configuration options
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32G0B1)  --->
    Bootloader offset (8KiB bootloader)  --->
    Clock Reference (8 MHz crystal)  --->
    Communication interface (USB to CAN bus bridge (USB on PA11/PA12))  --->
    CAN bus interface (CAN bus (on PD12/PD13))  --->
    USB ids  --->
(250000) CAN bus speed
()  GPIO pins to set at micro-controller startup

make clean
make

Put the Manta in DFU mode: Press equally Boot0 AND Reset (on the Manta board), relese Reset and then release Boot0
lsusb should give you:
Bus 001 Device 008: ID 0483:df11 STMicroelectronics STM Device in DFU Mode

sudo dfu-util -a 0 -D ~/CanBoot/out/canboot.bin --dfuse-address 0x08000000:force:leave -d 0483:df11

Press reset on Manta board
Then put the manta in DFU mode again

sudo dfu-util -a 0 -d 0483:df11 --dfuse-address 0x08002000 -D ~/klipper/out/klipper.bin
You should now see with lsusb:
Bus 001 Device 011: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter

If your other CAN0 network is correct you should now be up'n running. 
By the way, I choose baud 250000 since this was adviced by several sources 

You will not be able to update your Manta M8P with "python3 ~/CanBoot/scripts/flash_can.py -u UUID" since the CAN0 network goes down, so this have to be a manual operation with DFU mode as far as I have found out, for both CanBoot and Klipper. Other CAN devices you can update through "python3 ~/CanBoot/scripts/flash_can.py -u UUID"

NAPCAL, 2023-01-26:
If you remove :leave from the CanBoot dfu-util string it will stay in DFU mode.

eibo68, 2023-01-26:
@NAPCAL Thanks! I will try that now :-)

NAPCAL, 2023-01-26:
sudo dfu-util -a 0 -D ~/CanBoot/out/canboot.bin --dfuse-address 0x08000000:force:mass-erase -d 0483:df11

sudo dfu-util -a 0 -d 0483:df11 --dfuse-address 0x08002000:force:leave -D ~/klipper/out/klipper.bin

NAPCAL, 2023-01-26:
[image]

eibo68, 2023-01-26:
@NAPCAL Thanks again. It worked. With USB to CAN bridge I have to manually update the Manta M8P with dfu-util every time for klipper (and CanBoot). However my Hermit Crab CAN I use script:

# canboot-HermitCan.sh  -  Updating Hermit Crab CAN with new CanBoot
cd ~/CanBoot
make clean
cp .config.HermitCan .config
make
python3 ~/CanBoot/scripts/flash_can.py -f ~/CanBoot/out/deployer.bin -u 9056fdeabc65 

# HermitCan.sh  -  Update Hermit Crab CAN with new klipper
cd ~/klipper
make clean
cp .config.HermitCan .config
make
python3 ~/CanBoot/scripts/flash_can.py -u 9056fdeabc65

bddanford, 2023-01-27:
Thank you so much! I was able to get it working as well!

eibo68, 2023-01-27:
@bddanford Thanks for reporting the success back. Glad it helped :-)

airbornetrooper82573, 2023-01-28:
I have the v1.0 version without the can ports on the board. What do I need to change in the above settings to get this working?

NAPCAL, 2023-01-28:
V1.0 doesn't have a CAN bus transceiver on the board. Only two ways to add CAN bus U2C (USB to CAN bus) adapter OR RPi CAN bus hat (unique setup for CB1, simple for CM4)

eibo68, 2023-01-29:
@airbornetrooper82573 Or this cheap alternativ from aliexpress where you get 5v and ground from endstop and PB0 and PB1 running CAN bridge mode
https://www.aliexpress.com/item/1005002843325046.html?spm=a2g0o.order_list.order_list_main.56.7ad71802AHRH7Z

Jjustice1993, 2023-03-14:
So thanks to your guide i have this partially working, Thanks. I am having an issue though where anytime i say edit my config, have a firmware restart, etc, the can network seems to crash and will not reconnect to either my main MCU (manta m8) or secondary MCU (hermitcrab), even with a second firmware restart it wont connect again. I have todo a full host restart. Any ideas?

giantpt, 2023-04-03:
Hi all... this is a bit confusing.. I just bought a Manta M8P and an EBB36, is there a complete a fully working complete tutorial on how to make it working properly from A to Z?
Thanks

NAPCAL, 2023-04-03:
Not really for every setup.
1. RPi CAN bus hat.
2. USB to CAN bus adapter.
3. Klipper CAN bus bridge.

Please let us know what you are looking for guidance on.

giantpt, 2023-04-03:
Hi Napcal, What I needed was to have the EBB36 connected to the M8P via CAN but I managed to get it working, its all fine now. Thanks.

smplman, 2023-04-16:
@eibo68 thank you for your very detailed post. I was able to get my stuff working following your steps. My question is if we can't use CanBoot to update the FW is there any reason or benefit to using the MCU in CAN mode over serial?

I'm pretty new to CAN right now, but would it be possible to use the CM4 or CB1 to host the CAN network instead of the MCU so that we would be able to update the MCU firmware over CanBoot?

NAPCAL, 2023-04-16:
@smplman If you are talking about CAN over USB (not bridge mode) is good for boards with no CAN bus transceiver onboard but have to be paired with a U2C v1.1 or v2.1 with USB-A ports connected to CAN bus transceivers. (U2C V1.0 & v2.0 don't provide this functionality)

Note: For the M8P, M5P, M4P, and boards that have a CM4/CB1 onboard interface, the USB connection to the STM32 is not accessible due to it is a direct connection between the CM4/CB1 socket and the STM32, so the above option will not work on these boards.

WegnerDan, 2023-05-07:
For anyone that's looking for a way to update a Manta board while using it as a can bridge:
You can flash CanBoot with Communication interface "USB (on PA11/PA12)" instead of CAN onto the Manta board. 
This enables you to get the board to go into dfu mode over can, so you can update without accessing the board and setting boot jumpers or pressing reset buttons.

Updating consists of these two steps then:

`python3 flash_can.py -f ~/klipper/out/klipper.bin -u $mcu_id$`

This will fail, but leave the MCU in DFU mode and available through USB. Rerunning flash_can with the path to the serial device will successfully update and then reboot in CAN mode. 
In my case it looks like this:

`python3 flash_can.py -f ~/klipper/out/klipper.bin -d /dev/serial/by-id/usb-CanBoot_deviceblablabla`

Maybe there is a way to do this in a single step, but this is good enough for me. I have this info from a PDF that was shared on teamfdm forums: https://www.teamfdm.com/forums/topic/672-how-to-use-can-toolhead-boards-connected-directly-to-octopus-octopus-pro-on-canboot/?do=findComment&comment=9911