Arksine/katapult #65: Serial Flash -d  to MCU running klipper FW not dropping into Canboot FW.

Polar-Ted, 2023-02-03:
I'm trying to do a serial flash over USB to a board already loaded and running klipper and it fails to drop into the CanBoot loader when I run the script. .   I've done this in the past and it worked fine.. Tried it on 2 systems. My Pi4 and a Fly Pi.. both threw the same error.   If I double tap the reset button to kick it into the CanBoot loader, grab the CanBoot USB serial ID  and flash it will run fine..  See below for the errors. 

```
pi@pi4:~ $ python3 ~/CanBoot/scripts/flash_can.py -f ~/klipper/out/klipper.bin -d /dev/serial/by-id/usb-Klipper_stm32f407xx_20003B001547393436343430-if00
Attempting to connect to bootloader
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
```
this is truncated.. ask if you need more.

Arksine, 2023-02-03:
The script can't request the USB bootloader, that is up to the application (in this case Klipper).  This is typically done using the `make flash` command, presuming your device is supported.

Polar-Ted, 2023-02-03:
OK I got it now.. Closing this.. I'm an idiot..

this works
make flash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_stm32f072xb_240024001757425835303220-if00

closed 2023-02-24 by Arksine