Arksine/katapult #67: Error when flashing klipper on EBB36 v1.2 (STM32G0B1)

lukicsl, 2023-02-24:
I managed to flash CanBoot on the device and it is visible

```
~/CanBoot/scripts/flash_can.py -i can0 -q

Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 558da1510eb7, Application: CanBoot
Query Complete

```
I performed menueconfig in klipper and when I try to upload the klipper fw to the device I get following output:
```
python3 ~/CanBoot/scripts/flash_can.py -i can0 -f ~/klipper/out/klipper.bin -u 558da1510eb7

Sending bootloader jump command...
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 558da1510eb7, Application: CanBoot
Attempting to connect to bootloader
CanBoot Connected
Protocol Version: 1.0.0
Block Size: 64 bytes
Application Start: 0x8002000
MCU type: stm32g0b1xx
Verifying canbus connection
Flashing '/home/lukics/klipper/out/klipper.bin'...

[##################################################]

Write complete: 13 pages
Verifying (block count = 414)...

[ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
ERROR:root:Can Flash Error
Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 619, in main
    loop.run_until_complete(sock.run(intf, uuid, fpath, req_only))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 475, in run
    await flasher.verify_file()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 240, in verify_file
    resp = await self.send_command("REQUEST_BLOCK", payload)
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 186, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [REQUEST_BLOCK] to Can Device
```

I google'd around and did not get any hint on how to solve this issue

When querying the can again I can see klipper under the save uuid:

```
python3 ~/CanBoot/scripts/flash_can.py -i can0 -q

Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 558da1510eb7, Application: Klipper
Query Complete

```

Arksine, 2023-02-24:
Your can interface device is having trouble receiving data.  Is it a U2C v2?

lukicsl, 2023-02-24:
It is Waveshare RS485 CAN HAT

Arksine, 2023-02-24:
Make sure your device is configured with the [correct settings](https://www.waveshare.com/wiki/RS485_CAN_HAT) for its crystal.  You may also want to try reducing your can speed (this will require a rebuild and reflash of CanBoot).

The CAN Hats have a tendency to drop can frames when the throughput is high.  The gs_usb based devices (Canable, U2C, Klipper running USB-CAN bridge) are the most reliable.

lukicsl, 2023-02-24:
I ran at 500 000 I will reduce it to 250 000. crystal shall be ok at 12MHz

lukicsl, 2023-02-24:
I changed the speed to 250k.
Flashing is a succes, but still getting errors on the way:

```
Sending bootloader jump command...
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 558da1510eb7, Application: CanBoot
Attempting to connect to bootloader
CanBoot Connected
Protocol Version: 1.0.0
Block Size: 64 bytes
Application Start: 0x8002000
MCU type: stm32g0b1xx
Verifying canbus connection
Flashing '/home/lukics/klipper/out/klipper.bin'...

[##################################################]

Write complete: 13 pages
Verifying (block count = 414)...

[########ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
###ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/streams.py", line 632, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/lib/python3.9/asyncio/streams.py", line 517, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
[... 157 lines cut ...]
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/lukics/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
#####]

Verification Complete: SHA = 6CEB3ABB6F9B2D4D99FD3457D9239C56CDBFFBF2
CAN Flash Success
```

Arksine, 2023-02-24:
That is because the hat is still dropping frames, however it isn't doing it on 5 consecutive attempts (which is the maximum number of tries the `flash_can` script will attempt to send a command).  Its good enough to work, but be aware that it could also drop frames while running Klipper.

lukicsl, 2023-02-24:
In the future, I plan to use the can interface of octopus 1.1.
What speed is sufficient for klipper?

Arksine, 2023-02-24:
I don't really have a good answer for that, 250000 should be sufficient for one node.  There is also the issue of the max speed you reach, when depends on the length of your wiring.  You might get better answers on the Klipper Discourse or perhaps the Discord.

closed 2023-02-24 by lukicsl