Arksine/katapult #70: Could not find node corresponding to UUID starting with zero

odairgoncalvess, 2023-03-26:
Hi. I cannot flash the klipper.bin file using the flash_can.py script. It seems that the script cannot recognize the UUID of my EBB42 V1.2 BTT and omits the most significant digit - which is the number zero.

[image]

Arksine, 2023-03-26:
The leading zero is omitted from the log however it does not impact the internal representation of the UUID.  The issue seems to be that Klipper did not jump to CanBoot when requested.  This suggests that either CanBoot is not installed on the board or the version of Klipper on the board is not capable of performing the jump.

If you know CanBoot is installed you will need to manually enter the bootloader, presuming you enabled a method for doing so when you built CanBoot.  After flashing the latest version of Klipper future updates should be able to perform the jump.

NAPCAL, 2023-03-26:
@odairgoncalvess If you are using the BTT U2C [STM32G0B1] version for your CAN bus interface then this firmware may help.

**NOTE: this is not the firmware for the BTT U2C [STM32F072] version**

[G0B1_U2C_V2.zip](https://www.dropbox.com/s/dwk40um3wiz1pju/G0B1_U2C_V2.zip?dl=0)

odairgoncalvess, 2023-03-27:
@Arksine Thanks for the reply. The BTT42 board is with CanBoot and Klipper V0.11.0-86-g6026a99a. I will update manually by DFU and try in the next update.

odairgoncalvess, 2023-03-27:
@NAPCAL Thanks for the help. I'm not using the BTT U2C. I'm using the following hardware with CanBoot and CANbus: Orange Pi Zero2, MKS Monster8 V2 and BTT EBB42 V1.2.

odairgoncalvess, 2023-03-28:
It's work! Thank you!

I updated CanBoot v0.0.1-40 to v0.0.43 and it worked.

[image]