Arksine/katapult #71: Feature request: flash_can.py support for programmatically entering the bootloader via USB.

miklschmidt, 2023-04-05:
In the readme.md it states:
> Enter the bootloader. This will occur automatically if no program is detected. If you built CanBoot with an alternative method of entry you may use that. If upgrading from a currently flashed version of Klipper the flash_can.py script will command the device to enter the bootloader (currently for CAN devices only).

Is it possible and planned to implement this for USB as well? We've had quite a few bootloader related issues out of the box on some boards lately, so I would love to add CanBoot support to RatOS to facilitate recovery from those situations, and maybe move over to CanBoot completely depending on how that goes.

Laikulo, 2023-09-07:
Klipper supports this by sending the string ` \x1c Request Serial Bootloader!! ~` directly over serial. It actually needs to fail parsing as a command for this to work. It also needs to be it's own "block", so if the device is in use by other things, it may not be recognized. I may try to work up a PR for this later this week.

Laikulo, 2023-09-16:
While testing out a PR for this, it looks like Klipper only supported this on real serial ports. I've opened a PR to add it to USB-CDC virtual serial ports.
https://github.com/Klipper3d/klipper/pull/6340

Arksine, 2023-09-16:
FWIW, Klipper already has method for requesting the bootloader over CDC-ACM connections, and it has a script, `flash_usb.py`, that requests the bootloader.   Presumably it also has a script for requesting the bootloader over UART connections, however I'm unsure.

Application specific bootloader requests probably should not exist in `flashtool.py`, but rather in the application repository itself, since the application defines exactly how the bootloader should be requested.  The CAN bootloader request is a bit unique considering there are relatively few CAN bootloaders available.  In this case Katapult defines how the bootloader is requested and applications are expected to implement it.

Laikulo, 2023-09-18:
Ah, I was not aware that klipper actually implemented that in CDC, I've only had success with that method on arduino/FTDI like systems.

miklschmidt, 2024-04-13:
I just realized my dumbass didn't really understand how this works. Am i correct in assuming that klipper will boot into katapult (configured for USB communication) when attempting to flash with `make flash FLASH_DEVICE=whatever`?

miklschmidt, 2024-04-13:
I just verified that it does in fact work! And it was almost too easy. You may have just made my week without doing anything @Arksine. Closing! 👍

closed 2024-04-13 by miklschmidt