Arksine/katapult #72: EBB42 v1.2 does not work

Kaelum, 2023-04-10:
Using the steps defined in the document below for the EBB42 v1.2, it fails every time while downloading:

https://klipper.discourse.group/uploads/short-url/oKyxrVja7wW0J8toGB1pqfaGp4z.pdf

Clearing the memory works, but the following step to write does not.

NAPCAL, 2023-04-10:
Newer instructions from here.
[Octopus, EBB, CaBoot, & Klipper Can Bridge](https://www.dropbox.com/s/009tb5yiekiey1n/Octopus%20%26%20EBB%20CanBoot%2C%20Klipper%2C%20CAN%20bridge%282023-03-25%29.pdf?dl=0)

Kaelum, 2023-04-10:
The dfu-util command returns the following:

```
pi@ratos:~/klipper $ sudo dfu-util -a 0 -D ~/CanBoot/out/canboot.bin --dfuse-address 0x08000000:force:mass-erase -d 0483:df11
dfu-util 0.11-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
dfu-util: No DFU capable USB device available
```

Kaelum, 2023-04-10:
@NAPCAL THANKS!  I've spent all day on this was the STM32CubeProgrammer kept erroring out.  When moving to the Pi, I forgot to connect the EBB to the Pi and not my PC, but it's working great!  Now on to the next steps.  ;)

Kaelum, 2023-04-10:
@NAPCAL THANKS!  I've spent all day on this as the STM32CubeProgrammer kept erroring out.  When moving to the Pi with these new instructions, I forgot to connect the EBB to the Pi and not my PC, but it's working great now!  Now on to the next steps.  ;)

NAPCAL, 2023-04-10:
If this works out, please comment that it does and close out this issue.

Kaelum, 2023-04-10:
I'm on the RPi changes, and it's failing at step 6.  Nothing is listed.

NAPCAL, 2023-04-10:
Have you completed the wiring connections?

If you have recheck that CAN-H Octopus is connected to CAN-H EBB, and the same for CAN-L.

The EBB 120R jumper is in place, the EBB VBUS jumper is removed, and 24V is being provided to the EBB.

Kaelum, 2023-04-10:
Yeah, I've verified all of that.  What I am seeing after the reboot is that there aren't any serial devices.  The `/dev/serial` folder doesn't exist.

NAPCAL, 2023-04-10:
There will not be any serial devices since Klipper CAN bridge is running.

what is the output after lsusb command?

Kaelum, 2023-04-10:
pi@ratos:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

NAPCAL, 2023-04-10:
This is the Klipper CAN bridge.
The Octopus will only show as a CAN bus UUID.[image]

NAPCAL, 2023-04-10:
What is can0 showing in the output for ifconfig command?

It should be at the top, scroll up.

Kaelum, 2023-04-10:
can0 does not appear to be working:

```
pi@ratos:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.23.2.192  netmask 255.255.255.0  broadcast 10.23.2.255
        inet6 fd55:119f:3a3c:41ec:d3ef:5ce7:811c:7cf  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::f11:155f:3d1:bb0  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:63:ae:9f  txqueuelen 1000  (Ethernet)
        RX packets 665  bytes 116178 (113.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 207  bytes 33391 (32.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 1224  bytes 702698 (686.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1224  bytes 702698 (686.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.50.1  netmask 255.255.255.0  broadcast 192.168.50.255
        inet6 fe80::2c9:af9e:2489:f82f  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:63:ae:a0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 48  bytes 7856 (7.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

NAPCAL, 2023-04-10:
sudo nano /etc/network/interfaces.d/can0

Show what is in this file.

Kaelum, 2023-04-10:
```
pi@ratos:~ $ cat /etc/network/interfaces.d/can0
allow-hotplug can0
  iface can0 can static
  bitrate 500000
  up ifconfig $IFACE txqueuelen 1024
```

NAPCAL, 2023-04-10:
Exit nano
Reboot
Check for can0 with ifconfig

Kaelum, 2023-04-10:
It's still not working:

```
pi@ratos:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.23.2.192  netmask 255.255.255.0  broadcast 10.23.2.255
        inet6 fe80::f11:155f:3d1:bb0  prefixlen 64  scopeid 0x20<link>
        inet6 fd55:119f:3a3c:41ec:d3ef:5ce7:811c:7cf  prefixlen 64  scopeid 0x0<global>
        ether dc:a6:32:63:ae:9f  txqueuelen 1000  (Ethernet)
        RX packets 164  bytes 36666 (35.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 155  bytes 25116 (24.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 239  bytes 248362 (242.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 239  bytes 248362 (242.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.50.1  netmask 255.255.255.0  broadcast 192.168.50.255
        inet6 fe80::2c9:af9e:2489:f82f  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:63:ae:a0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 35  bytes 4651 (4.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

NAPCAL, 2023-04-10:
sudo ip link set up can0 type can bitrate 500000

Kaelum, 2023-04-10:
Ok, if I query for `can0` specifically, I get:

```
pi@ratos:~ $ ifconfig can0
can0: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

NAPCAL, 2023-04-10:
It looks like it is not up.

sudo ip link set up can0 type can bitrate 500000

Kaelum, 2023-04-10:
Even after running that command and rebooting, it still shows:

```
can0: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Kaelum, 2023-04-10:
It's like the firmware on the Octopus Pro 1.0 is even running.

NAPCAL, 2023-04-10:
The Flags before 128<NOARP>, now 193<UP,RUNNING,NOARP>

Now try step 6

Kaelum, 2023-04-10:
Actually, `ifconfig can0` is now not returning anything.

```
pi@ratos:~ $ ifconfig can0
can0: error fetching interface information: Device not found
```

Kaelum, 2023-04-10:
I think I see what happened.  Because we were last working on the Octopus Pro, that is the firmware that it sent to the EBB.  Can this be fixed?  Do I just so the EBB42 steps all over again?

NAPCAL, 2023-04-10:
sudo nano /etc/rc.local

Line before exit 0

sudo ip link set up can0 type can bitrate 500000
sudo ifconfig can0 txqueuelen 2048

Save and reboot

NAPCAL, 2023-04-10:
> I think I see what happened.  Because we were last working on the Octopus Pro, that is the firmware that it sent to the EBB.  Can this be fixed?  Do I just so the EBB42 steps all over again?

Start over with the EBB steps.

Kaelum, 2023-04-10:
It looks like I'm getting much closer.  The current issue is that the RatOS configuration has entries for `baud` and `serial`, which Klipper is complaining about.  This is my first time using Klipper.  I know it allows overrides to replace previously set values, but is there a way to remove/delete a key?  If not, I'm going to have to duplicate a lot of files to do this all manually.

NAPCAL, 2023-04-10:
Well you have stepped way from what I know.

Kaelum, 2023-04-10:
@NAPCAL well, Thank You!

I need to figure out what is going on with the Klipper configuration, but I think the Octopus Pro and the EBB might be working.  I'll let you know either way, once I fix this current issue.

Best!

Kaelum, 2023-04-10:
@NAPCAL ok, it is partially working, but it seems that `can0` is flakey.  The `/etc/rc.local` change doesn't seem work very often.  If I manually enter the following after the system has started, it seems to work:
```
sudo ip link set up can0 type can bitrate 500000
sudo ifconfig can0 txqueuelen 2048
```
In fact the fan on my hotend is on for the first time.

NAPCAL, 2023-04-10:
Do a web search for Debin can bus not starting up at boot.

See what works for others and give them a try.

If the one you try doesn't work then remove it and try the next.

Kaelum, 2023-04-10:
Ok, I've learned a lot, and figured out a lot since last night.  The biggest mind blower is that it doesn't look like **CanBoot** is needed at all.  Klipper has everything that is needs out of the box to fully support CAN in this configuration, but the documentation is extremely bad, and mostly missing, on the subject.  I still need to try it on the EBB42, but I completely removed it from the Octopus Pro v1.0 and restored the original boot.  As for `can0`, this seems to work for startup:
```
allow-hotplug can0
iface can0 can manual
    bitrate 500000
    up ifconfig $IFACE txqueuelen 2048
    pre-up ip link set up can0 type can bitrate 500000
```
However, the **FIRMWARE_RESTART** command disables `can0`, and I'm still looking for a work-around for that.  Once I'm done with my EBB42 tests, I'll post the results at that point.  If I don't have a resolution for **FIRMWARE_RESTART** by then, I'm post another follow up once I do.

NAPCAL, 2023-04-11:
If you don't use CanBoot then your stuck with having to use an SD card to update Klipper.

Your printer.cfg and EBB.cfg should have restart_method: commented out since this is more for USB connections (this doesn't include CAN bridge)

Kaelum, 2023-04-13:
Hey @NAPCAL,

Sorry it took so long for me to respond, but I've been a little busy with work.  What you state above isn't exactly accurate.  Both the Octopus Pro and the EBB42 have DFU modes, which require human intervention to activate.  Bypassing DFU is a severe security risk.  It would allow anyone who has access to your network, including hackers, to change the firmware on your devices and should never be allowed under any circumstance.  Any tool that is used to update firmware **MUST** work within the confines of DFU in order to safeguard your devices.

Using the normal boot code on the Octopus Pro in no way interferes with direct programmatic updates, as long as those updates are designed to work with DFU.  Updating the EBB42 might be more of a challenge, as DFU requires that you use the USB-C connector, but that can easily be worked around by temporarily using a USB-A to USC-C cable from the Raspberry Pi to the EBB42, and placing the EBB42 in DFU mode.  You might consider this cumbersome, but anyone who is security conscious would not.

Best

closed 2023-04-13 by Kaelum

NAPCAL, 2023-04-13:
@Kaelum There maybe a misunderstanding about STM and there DFU (Device Firmware Update) mode that allows you to program the STM processors. There are two ways to do this one is the method described and the second is to use a STMLink device. The stock bootloader code by BTT requires the use of a SD card to get the firmware loaded starting at the location that the bootloader jumps to when it is done it's thing.

CanBoot is a replacement bootloader that does the same thing but via USB, UART, or CAN bus without the need for an SD card. So updating Klipper firmware can be done without transferring the klipper.bin to a PC then to an SD card, renaming klipper.bin to firmware.bin then ejecting the SD card, placing that card in to the printer board, resetting or powering up the board, guess if you gave it enough time, removing the SD card, placing back into a PC to see if it got renamed to firmware.CUR, then you are still questioning if it is running new firmware.

With CanBoot you get confirmation of the process and the update took.

Kaelum, 2023-04-13:
@NAPCAL there is no misunderstanding.  With the normal bootloader I have the following options:
1. Using an SD card to load new firmware.
1. Placing the jumper on the Octopus Pro and resetting it to enter DFU mode, from with I can update it via the connection to the Raspberry Pi.
These are the only options that I want, and they are the only options that I need.  If Klipper currently does not fully support this, that is on Klipper, as there is no reason at all that Klipper can't support this without the use of CanBoot.

Kaelum, 2023-06-10:
@NAPCAL the latest code now works w/o any issues with CAN.  You do need to change your instructions a bit though.  The `can0` file that I'm using has:

```
allow-hotplug can0
iface can0 can static
    bitrate 500000
    up ifconfig $IFACE txqueuelen 2048
    post-up /sbin/ip link set up $IFACE type can bitrate 500000
```

For Canboot, the make menu config needs to be this:

[image]

For Klipper, the make menuconfig need to be this:

[image]

Then to flash the Octopus board, the following commands need to be run while in DFU mode:

```
dfu-util -a 0 -D ~/CanBoot/out/canboot.bin -s 0x08000000:mass-erase:force

dfu-util -a 0 -D ~/klipper/out/klipper.bin -s 0x08008000:leave
```

Once everything is rebooted, CAN will just work.  However, I don't know how to get the UUID of the board when doing this.  I already had the UUID, so it worked, but I can't figure out how to get the UUID now.

NAPCAL, 2023-06-10:
~/CanBoot/scripts/flash_can.py -q

If using can1, then -i can1 to the command above and add canbus_interface: can1 to the cfg file for Klipper to use the correct interface.

P.S. can0 is not needed since most commands and cfg default to can0.

The other changes you have listed are not required or don't affect the CAN bus and tend only to be specific to your OS, hardware, or configuration.

Kaelum, 2023-06-10:
`pythin3 ~/CanBoot/scripts/flash_can.py -q` doesn't return anything, so there may be a bug in the script.  Anywho, w/o the changes described the Raspberry Pi will lose communication with the Octopus (Pro) and `can0` will not be available.  There will also be a timing bug with **FIRMWARE RESTART**, which requires that a **RESTART** be issued shortly after it.  In order to ensure that everything works as it is supposed to, I took pieces from what you presented and the incomplete work from `docgalaxyblock/KlipperMisc` (which fixes the `can0` and **FIRMWARE RESTART** issues) in order to come up with a full solution.

Best...

NAPCAL, 2023-06-10:
@Kaelum 

For Klipper to release the CAN bus devices it has control over you need to stop the klipper service before running flash_can.py then start it back up after.

Kaelum, 2023-06-11:
Ok, stopping Klipper didn't work, but commenting out the UUID and restarting did.  Good to know.  Thanks