Arksine/katapult #75: Can't flash klipper on canbus

jwhite, 2023-05-24:
I'm getting this error when I flash klipper with a 16k offset using the canbus.

The board has a fast flashing led (very fast) and doesn't seem to work after the flash.

Flashing klipper without canboot work fine.

pi@voron:~/CanBoot/scripts $ python3 flash_can.py -i can0 -f ~/klipper/out/klipper.bin -u 3799962ca524
Sending bootloader jump command...
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 3799962ca524, Application: CanBoot
Attempting to connect to bootloader
CanBoot Connected
Protocol Version: 1.0.0
Block Size: 64 bytes
Application Start: 0x10004000
MCU type: rp2040
Verifying canbus connection
Flashing '/home/pi/klipper/out/klipper.bin'...

[###################################ERROR:root:Can Flash Error
Traceback (most recent call last):
  File "flash_can.py", line 483, in run
    await flasher.send_file()
  File "flash_can.py", line 214, in send_file
    resp = await self.send_command('SEND_BLOCK', prefix + buf)
  File "flash_can.py", line 194, in send_command
    % (cmdname))
FlashCanError: Error sending command [SEND_BLOCK] to Can Device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "flash_can.py", line 628, in main
    loop.run_until_complete(sock.run(intf, uuid, fpath, req_only))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "flash_can.py", line 489, in run
    await flasher.finish()
  File "flash_can.py", line 272, in finish
    await self.send_command("COMPLETE")
  File "flash_can.py", line 194, in send_command
    % (cmdname))
FlashCanError: Error sending command [COMPLETE] to Can Device

jwhite, 2023-05-25:
SB2040 I flashed previously with 16k offset worked.  This one doesn't.  No idea why.

NAPCAL, 2023-05-25:
SB2040 is not a BTT product; the BTT EBB SB2209 & SB2240 are BTT products.

The boot offsets I listed are for BTT products; I don't have or provide guides on other manufacturers' products.

treaves, 2024-01-22:
@NAPCAL yes, BTT does make a CAN EBB that uses the SB2040 in addition to the other two.

NAPCAL, 2024-01-22:
@treaves You are mixing up drivers with processors.

The SB EBB are 2209 or 2240
The SB2209 can come with a STM32G0Bx or RP2040 there is no SB2040 Processor.

There is a https://github.com/Mellow-3D/Fly-SB2040 that is not a BTT product.

treaves, 2024-01-22:
I see. Thanks.

icase81, 2024-06-23:
I'm getting this same error on an EBB2209 RP2040.

`icase@kobra2neo:~/klipper $ python3 ~/katapult/scripts/flashtool.py -i can0 -u 9939ed493165 -f ~/klipper/out/klipper.bin
Sending bootloader jump command...
Resetting all bootloader node IDs...
Attempting to connect to bootloader
Katapult Connected
Protocol Version: 1.0.0
Block Size: 64 bytes
Application Start: 0x10004000
MCU type: rp2040
Verifying canbus connection
Flashing '/home/icase/klipper/out/klipper.bin'...

[####################ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/icase/katapult/scripts/flashtool.py", line 481, in run
    await flasher.send_file()
  File "/home/icase/katapult/scripts/flashtool.py", line 217, in send_file
    resp = await self.send_command('SEND_BLOCK', prefix + buf)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/icase/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [SEND_BLOCK] to Can Device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/icase/katapult/scripts/flashtool.py", line 626, in main
    loop.run_until_complete(sock.run(intf, uuid, fpath, req_only))
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/icase/katapult/scripts/flashtool.py", line 487, in run
    await flasher.finish()
  File "/home/icase/katapult/scripts/flashtool.py", line 275, in finish
    await self.send_command("COMPLETE")
  File "/home/icase/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [COMPLETE] to Can Device
icase@kobra2neo:~/klipper $ 
`