Arksine/katapult #80: Klipper can't connect to mcu via canbus after Raspbian 11 (bullseye) update.

Flob74, 2023-07-18:
Hi,

I updated Raspbian 10 (buster) to Raspbian 11 (bullseye), for crowsnest compatibilities issues. But now Klipper is unable to connect to can mcu.
I'am using a RPi 3B, an octopus v1.1 (STM32F446) and a SB2040 (RP2040). Each MCU is connected via canbus.

> pi@voron24:~ $ ifconfig
can0: flags=193<UP,RUNNING,NOARP>  mtu 16 unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1024
pi@voron24:~/canboot/scripts $ python3 flash_can.py -i can0 -q
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 3f56a5b6c36c, Application: Klipper
Detected UUID: 2d70e57a001c, Application: Klipper
Query Complete

Everything is up-to-date in the mainsail update manager or in kiauh. machine.cfg is included in printer.cfg and containt both mcu parameters and adresses. But mainsail still can't connect to klipper.
I even tryed to reflash canboot on the octopus board. No success.
[moonraker.log](https://github.com/Arksine/CanBoot/files/12088987/moonraker.log)
[klippy.log.json.txt](https://github.com/Arksine/CanBoot/files/12088996/klippy.log.json.txt) (I add ".txt" for uploading)
[machine.cfg.txt](https://github.com/Arksine/CanBoot/files/12089042/machine.cfg.txt) (which is imported in printer.cfg)
[printer.cfg.txt](https://github.com/Arksine/CanBoot/files/12089056/printer.cfg.txt)

If I missed something, surely someone else will too. So, thank you for your help.

Mainsail config and errors :
[image]

Flob74, 2023-07-30:
Ok. I delete everything. I reflashed the Raspberry PI with MainsailOS on Raspbian 11 (bullseye). I manually reinstall every plugins I had. I reflashed my MCUs canboots. And everything works again. 
So I can't tell where the error came from.

Sorry for the inconvenience.

Surion79, 2023-11-21:
you could close your own issue ;)