Arksine/katapult #81: flashtool.py cannot find nodes

Fuergrissa1, 2023-07-31:
I am following this guide: https://docs.meteyou.wtf/

Building and flashing bootloaders (BTT candleLight_fw for U2C 2.1, katapult for EBB36 1.2 and 2x EBB42 1.2) using dfu-util over USB seems to go fine. Building klipper again goes fine, but to flash over CAN fails because no devices are detected on the CAN network. 

`pi@fluiddpi:~ $ python3 ~/katapult/scripts/flashtool.py -i can0 -q
Resetting all bootloader node IDs...
Checking for Katapult nodes...
Query Complete
pi@fluiddpi:~ $`

I can go ahead and flash klipper over USB, so I have tried that, but when I reconnect for CAN communication the same problem persists.

I have tried re flashing all boards with setting CAN speed to 500k and 1M, which has not helped.

I am trying this with only one EBB board at a time, so my connections are Pi to U2C (USB) U2C to EBB (CAN). I am using the screw terminals on the U2C and I have quadruple checked the wiring- Can-H to Can-H, CAN-L to CAN-L, +24 to +24, and GND to GND. Since I am only testing with one node at a time I have 120R jumpers on both the U2C and the EBB. Measuring CAN-H to CAN-L with power off is 60 Ohms.

Can you recommend any further troubleshooting steps?

Arksine, 2023-07-31:
According to the linked documentation these boards have user LEDs and the recommended configuration has them configured in the Katapult build.  When Katapult is active and ready to accept commands the led will blink slowly.  My first suggestion is to use the LED to verify Katapult is active.

If the led is not blinking try manually requesting the bootloader by pressing the reset button twice quickly.  If the reset button is small and/or difficult to access you may need a small tool to perform the double reset.

If Katapult is active yet doesn't show up when queried then I would take a hard look at the U2C v2.1.  Meteyou's guide doesn't seem to address how to bring up the CAN network interface, however it appears that you do have it up.  You will need to make sure the bitrate of the can socket matches the bitrate configured in Katapult/Klipper.  You can check it with the following command:

```
ip -det addr
```

Fuergrissa1, 2023-08-02:
The LED does blink slowly after flashing katapult.

```
pi@fluiddpi:~/katapult $ ip -det addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00 promiscuity 0 minmtu 0 maxmtu 0 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether e4:5f:01:4d:aa:45 brd ff:ff:ff:ff:ff:ff promiscuity 0 minmtu 68 maxmtu 1500 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
    inet 192.168.68.85/22 brd 192.168.71.255 scope global dynamic noprefixroute wlan0
       valid_lft 6527sec preferred_lft 5627sec
    inet6 fe80::7952:71a1:f537:e92a/64 scope link
       valid_lft forever preferred_lft forever
4: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0
    can state STOPPED restart-ms 0
          gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
          clock 64000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

So it looks like the CAN network is DOWN. After googling I tried
```
pi@fluiddpi:~ $ sudo ip link set can0 type can bitrate 500000
pi@fluiddpi:~ $ sudo ip link set up can0
pi@fluiddpi:~ $ ip -det addr
4: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0
    can state ERROR-ACTIVE restart-ms 0
          bitrate 500000 sample-point 0.875
          tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
          gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
          clock 64000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

which looks good, so checking again for CAN nodes:
```
pi@fluiddpi:~/katapult $ python3 ~/katapult/scripts/flashtool.py -i can0 -q
Resetting all bootloader node IDs...
Checking for Katapult nodes...
Detected UUID: c5519d4f29c6, Application: Katapult
Query Complete
```

Hooray! but how can I avoid having to manually bring up the CAN network? meteyou's guide has me create /etc/network/interfaces.d/can0 and fill it with:
```
allow-hotplug can0
iface can0 can static
    bitrate 500000
    up ifconfig $IFACE txqueuelen 128
```

so I can clearly see that it contains the words "bitrate 500000", and "up" is this not being honored or am I off track here?

Also, thank you so much!

Arksine, 2023-08-06:
AFAIK that should bring up the can network when the usb device is detected.  My understanding is that fluiddpi is deprecated and 'has issues', although I don't know if they are in any way related to the canbus.

Fuergrissa1, 2023-08-12:
You are right, I reflashed Pi OS lite with the the Pi flash tool and used kiauh to handle all of the installs. after updating /etc/network/interfaces.d/can0 according to the meeteyou guide and re-cloning katapult I can now detect Katapult nodes without bringing up can0 manually.

Thanks!

closed 2023-08-12 by Fuergrissa1