Arksine/katapult #83: CanBoot / katapult has never worked

Kaelum, 2023-08-13:
I tried another update today, as `klipper` required that the MB be flashed, and I just realized that this has never worked.  I can install `katapult` and `klipper` via DFU mode with the following:

```
dfu-util -a 0 -D ~/katapult/out/katapult.bin -s 0x08000000:mass-erase:force -d 0483:df11
dfu-util -a 0 -D ~/klipper/out/klipper.bin -s 0x08008000:force:leave -d 0483:df11
```

However, I can NEVER do anything with `flashtool.py` (previously `flash_can.py`).  The following command to use `katapult` to flash the MB has never worked:

```
python3 ~/katapult/scripts/flashtool.py -i can0 -f ~/klipper/out/klipper.bin -u 28b14df6273c
```

After stopping `klipper` and executing the above, the result is always:

```
python3 ~/katapult/scripts/flashtool.py -i can0 -f ~/klipper/out/klipper.bin -u 28b14df6273c
Sending bootloader jump command...
Resetting all bootloader node IDs...
Checking for Katapult nodes...
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/pi/katapult/scripts/flashtool.py", line 628, in main
    loop.run_until_complete(sock.run(intf, uuid, fpath, req_only))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/katapult/scripts/flashtool.py", line 474, in run
    raise FlashCanError(
FlashCanError: Unable to find node matching UUID: 28b14df6273c
```

And `28b14df6273c` is the UUID of my Octopus Pro MB.

Kaelum, 2023-08-13:
On a side note, I have the CAN of the Octopus Pro 1.0 connected to an EBB42, and CanBoot / katapult is able to flash that board w/o any issues.

NAPCAL, 2023-08-13:
@Kaelum If you are using Klipper with CAN bus bridge mode, then try this. Katapult set to communicate over USB with double click reset to jump to update mode, and Klipper set to communicate via Bridge mode.

When trying to update Klipper via CAN bus, it can get a little tricky since it is the interface to the bus.

With Katapult on the USB instead of the CAN bus, just double-click the MCU reset; easy to wire up if you don't have a display connected to EXP1 & EXP2, then load the new Klipper or Deployer (Katapult) via USB.

Kaelum, 2023-08-14:
@NAPCAL if I double click the reset button to put the Octopus Pro into bootloader mode, it disappears from USB.  Before:

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 003: ID 1bcf:28c4 Sunplus Innovation Technology Inc. 3DO NOZZLE CAMERA 4K
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

After:

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1bcf:28c4 Sunplus Innovation Technology Inc. 3DO NOZZLE CAMERA 4K
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

NAPCAL, 2023-08-14:
@Kaelum If it was set for USB then
ls /dev/serial/by-id

You should see one starting with usb-Katapult

If it was set for CAN bus then Klipper is not running so no bridge so no CAN bus.

Kaelum, 2023-08-14:
The instructions said to set it up like this:

[image]

Are you saying to change it to this?

[image]

I seem to remember that CAN won't work at all if I do that.

NAPCAL, 2023-08-14:
@Kaelum Give it a try; Katapult doesn't monitor the USB port until it is dropped into firmware loading.

The settings you have will create the deployer.bin if that is the step you are on.

NAPCAL, 2023-08-14:
After getting Klipper with Bridge mode loaded you will see this after doing a lsusb.
[image]

Kaelum, 2023-08-14:
@NAPCAL Ok. I changed the Katapult configuration for the Octopus Pro to USB, and Klipper is still working as it was.  When I double click the reset button now, the following serial device appears:

```
usb-katapult_stm32f429xx_1B0033000850314B33323220-if00
```

Do I use the following to flash it?

```
python3 ~/katapult/scripts/flashtool.py -f ~/klipper/out/klipper.bin -d /dev/serial/by-id/usb-katapult_stm32f429xx_1B0033000850314B33323220-if00
```

NAPCAL, 2023-08-14:
@Kaelum 

That will work if you have already installed the Python serial.

If not run this:
pip3 install pyserial

Kaelum, 2023-08-14:
Ok, cool.  I tried it, and now it's working.  Thanks!  Now I can patch both boards much easier.  I didn't mind the DFU mode until I was forced to mount fans to cool my Octupus Pro, and removing them is a major pain to enable DFU mode.

The documentation is very confusing.  It would be nice if they had an example page to show this.

NAPCAL, 2023-08-14:
@Kaelum 

Good,

It is a wrinkle that is only introduced when using the Klipper Bridge.

On the tool head board, you will not have issues with Katapult & Kipper communicating via CAN bus.

Kaelum, 2023-08-20:
@NAPCAL well, I thought this was working, but it isn't.  While patching the Octopus Pro is now possible after double clicking the reset button, patching the EBB42 is no longer possible.  Why can't this just work?  This is so stupid.

Arksine, 2024-01-03:
I'm going to close this as the discussion seems to have concluded.  As has been mentioned, devices in USB-to-CAN bridge mode can't be flashed over the canbus.  They are connected via USB and thus need to be flashed over USB via CDC-ACM.

The other thing to keep in mind when using usb-to-can bridge mode is that you should flash/update your nodes first.  When you reset the bridge to enter the bootloader the CAN network will go down, making the nodes unreachable.

closed 2024-01-03 by Arksine