Arksine/katapult #87: [Enhancement] firmware: Support reporting the current version of Katapult.

Laikulo, 2023-09-07:
Add a command (or extend CONNECT) to report the version of canboot (possibly from `git describe` or a `.version` file if not in a git repo at build time). This would allow for software to query and update canboot (via deployer or other methods) conditionally.

gpread, 2023-09-10:
The ability to receive Katapult  version info as well as UUID in a query response would be very helpful.  With multiple versions and since AFAIK there is currently no easy  automatic Katapult version updater (and perhaps people would not be easily convinced to use one for updates anyway) it is valuable to know definitively if you need to update and how many revisions you may be skipping when deploying the most current one and/or which version you would be wanting to revert to if you found an incompatibility/problem with newer one.

krautech, 2024-07-21:
This functionality would also be greatly appreciate for everything already mentioned. I'll look into this myself but no guarantees. @Arksine anywhere I should be looking in specific to pulling the current Katapult version on the device?

kdomanski, 2024-08-03:
Hey, I've been looking into this a while ago and the update to the thread reminded me to write down the takeaways.

The only workable solution IMO seems to be passing data to the Klipper payload similar to how bootloaders pass parameters (so-called "atags") to the Linux kernel. It seems to work by reserving a small piece of memory above the stack and passing its address through a register before jumping to the kernel. The address contains a list of structs. Then Klipper would have to pick up the address, validate the data and read it.

See:
- https://github.com/torvalds/linux/blob/661e504db04c6b7278737ee3a9116738536b4ed4/arch/arm/kernel/head-common.S#L142
- https://github.com/torvalds/linux/blob/661e504db04c6b7278737ee3a9116738536b4ed4/arch/arm/kernel/setup.c#L1102

Arksine, 2024-08-06:
FWIW, I have an implementation of this that I intend to push soon.  As requested, it uses `git describe` and falls back on a `.version` file in the katapult folder if its not a git repo.

There are several approaches that can be taken to accomplish this, I used Kconfig.  By doing so the version is defined in `autoconf.h`, making the change minimal.  This means that its a good idea to clean before building a new version, since we need to make sure that `autoconf.h` is regenerated before the build.  Failing to do so could result in a build containing stale version info.

krautech, 2024-08-06:
Excellent news!

That's roughly what I came up with temporarily but my version involved using json and having version information hosted somewhere (git etc)

Yours is much more elegant 👍🏻

Arksine, 2024-08-06:
I have added this commit to PR #126.  If some of you want to test it before I merge feel free to do so.

The `flashtool` script should still work for devices flashed with previous versions of Katapult, the software version should return as "?".   I have done some initial testing, but would still like to test all of the interfaces before I merge.

The output should look like the following:
```
Flashing Serial Device /dev/ttyACM0, baud 250000
Attempting to connect to bootloader
Katapult Connected
Software Version: v0.0.1-68-gb768c12
Protocol Version: 1.1.0
Block Size: 64 bytes
Application Start: 0x8002000
MCU type: stm32f103xe
Flashing '/home/eric/klipper/out/klipper.bin'...
```

krautech, 2024-08-07:
Ill run through this a bunch today and tomorrow and provide feedback 👍🏻

krautech, 2024-08-07:
> I have added this commit to PR #126. If some of you want to test it before I merge feel free to do so.
>
> The `flashtool` script should still work for devices flashed with previous versions of [...]

Is working for me so far :D

closed 2024-09-03 by Arksine