Arksine/katapult #91: Hot-Plug ebb36 and octopus 1.1 466

islonina, 2023-10-11:
What im missing. EBB36 and octopus on CAN all works fine but i need to manualy up the can0. I have file in interfaces for hot-plug:
allow-hotplug can0
iface can0 can static
    bitrate 500000
    up ifconfig $IFACE txqueuelen 128

And still need to use :
sudo ip link set can0 type can bitrate 500000
sudo ip link set up can0

or in single line:
/sbin/ip link set up can0 type can bitrate 50000

kdomanski, 2023-10-12:
You're missing the line `auto can0`.

NAPCAL, 2023-10-12:
@kdomanski When using allow-hotplug can0 instead of auto can0 

allow-hotplug basically means for OS: this interfaces is dynamic, manage it on various condition changes.

auto basically is telling to OS: bing up this interface with provided configuration during boot time or interface link up event.