Arksine/katapult #92: STM32G0B1 via STM32CubeProgrammer fails

Collapsednut, 2023-10-14:
Unable to flash via STM32Cube. I am able to flash klipper via STM32Cube, but katapult fails. Reference attached screenshot. 
[image]

tkff, 2024-01-19:
Same problem here.

Arksine, 2024-01-20:
I'll look into it.  Until a couple of days ago I did not have this specific MCU to test with directly.  I suspect it is related to the build environment, so some additional details will be helpful:

1) The Katapult build configuration
2) The Linux distro and version you are building on (ie: Debain Bookworm, Ubuntu 22.04, etc)
3) The version of your toolkit.  You can get it with the following command:
```
arm-none-eabi-gcc --version
```