Arksine/katapult #93: Coexisting canboot.bin and katapult.bin in katapult/out folder

Phil1988, 2023-10-14:
Hey,

I am following the voron canbus guide (see [here](https://github.com/Esoterical/voron_canbus/tree/main/mainboard_flashing)) and want to flash the katapult (formerly CanBoot) bootloader to the mainboard.

The tutorial says I should flash like
`dfu-util -R -a 0 -s 0x08000000:force:mass-erase:leave -D ~/CanBoot/out/canboot.bin -d 0483:df11`
and as the name meanwhile changed to katapult, I had a closer look.

In the out folder there is a **canboot.bin and a katapult.bin**:
[image]
Which is which and what should I use?
I think that katapult.bin is right, but there might be a reason, why canboot.bin is there and it may be different.

Thanks for your help :)

NAPCAL, 2023-10-14:
The changes will be:
From
~/CanBoot
To
~/katapult

From
~/CanBoot/out/canboot.bin
To
~/katapult/out/katapult.bin

From
~/CanBoot/scripts/flash_can.py
To
~/katapult/scripts/flashtool.py

All of the switches like -q -u are the same.

I haven't had the free time to update my guides, but it is on my list.

Phil1988, 2023-10-14:
Thanks @NAPCAL 

I know that, but my question is, why there are two -.bin files.

Are they currently similar/a copy to have backwards compatibility in the transition phase?
If not, what is different between canboot.bin and a katapult.bin?

Thanks

NAPCAL, 2023-10-14:
They did this so users accustomed to CanBoot could transition over to Katapult.

They also created a symbolic link from flash_can.py to flashtool.py

If you currently have CanBoot firmware installed, you can use the deployer option and flashtool.py to update to Katapult on that device.

python3 ~/katapult/scripts/flashtool -f ~/katapult/out/deployer.bin -u <UUID>

Phil1988, 2023-10-14:
Ok, so it is a the same file.

Then I'll close this.

Would be nice if this could be documented somewhere of if there is a general "how to" section somewhere.
I might not be the only one to appreciate that :)

closed 2023-10-14 by Phil1988