Arksine/katapult #95: Clarification of "Application start offset"

DrKlipper, 2023-11-26:
Hi !

Can anyone give me some explanation what "Application start offset" is all about? 
I thought that katapult replaces the original bootloader and that´s it. 
You have katapult & klipper within your flash memory and all is fine. 

But I wonder how "Application start offset" is used for? 
Is it to preserve the original bootloader? So in the end you have Original Bootloader, katapult and klipper within the flash memory? 

Would be great if anyone can explain this parameter a little bit in detail. 
Ths alot !

Dominik

NAPCAL, 2023-11-26:
The Application offset is an address location that Katapult will jump to after Katapult is done with its code setup and execution.

Typically this will be the start location of firmware code like Klipper or Marlin.

Arksine, 2023-11-29:
No, it isn't possible.  Katapult must be uploaded to the first address in flash memory, which is where the stock bootloader resides.

DrKlipper, 2023-11-29:
Hi Eric, 
thx for that response.

closed 2023-11-29 by DrKlipper