Arksine/katapult #96: Flashing RP2040 - No rule to make target 'flash'

alextac98, 2023-11-30:
Hi all,

Just bought a new BTT SB2209 with an RP2040 and wanted to install Katapult to be able to flash it over CAN. I ran through the instructions until it was time to flash to the board, and I was getting this error (after running make first):
```
$ make flash FLASH_DEVICE=2e8a:0003
make: *** No rule to make target 'flash'.  Stop
```

Looking at the Makefile, I didn't see a `flash` target, but I also never saw one in a quick history search in the file. Am I missing something? Is there a dependency needed that I don't have? I am doing this all on a Raspberry Pi, so is there an incompatability issue with arm chips?

Thanks!

Arksine, 2023-12-04:
Each platform has its own Makefile, the targets are located in there (ie: `src/rp2040/Makefile`).  To flash Katapult on an RP2040  you run `make flash`.  There is no `FLASH_DEVICE` option.

TimGehres, 2023-12-04:
Thanks, that solved the problem. It was a long day.

Burnsian, 2024-05-18:
I got excited when I read this thinking it might solve my issue, but sadly no. I'm the noobest of noobs, but I'm following BigTreeTech's manual on the SB2209 CAN (RP2040), page 29. After setting the SB2209 into 'boot' mode, querying DFU device ID, I came up with 2e8a:0003, identical to the ID shown in the guide. Then I entered the next command in the manual verbatim, "make flash FLASH_DEVICE=2e8a:0003", then I get the same error code "no rule to make target 'flash'". It may be that your answer above already should have given me a clue, but sadly I need a little more hand holding in order to understand what to do. Any help would be appreciated!

omgsideburns, 2024-12-28:
I stumbled across this while running in to the same error flashing an ADXL.. My issue was I was in the wrong directory because I had rebooted after generating the firmware. Hope that helps someone.

JimHejmanowski, 2025-01-06:
Thanks omgsideburns! You undoubtedly just saved me a few hours. I too was in the wrong directory.

snipervi2000, 2025-02-16:
For those who may stumble across this post and found the comments cryptic, you need to make sure that you are in the katapult directory. 

cd ~/katapult
make flash FLASH_DEVICE=2e8a:0003