Arksine/katapult #99: EBB SB2209 CAN RP2040 - No RP2040 Option in Katapult Configuration

TimGehres, 2023-12-04:
I ran the following but the Katapult Configuration doesn't list the Raspberry Pi RP2040 option, it only  lists LPC1786 options. 

Am I missing something, like not getting the correct git files? 

I copied and ran this out of **https://github.com/Arksine/katapult** website.

  git clone https://github.com/Arksine/katapult
  cd katapult
  make menuconfig
 
Here is a screenshot of the Configuration menu that I get when running make menuconfig
****
[image]

So, this is what it should look like:

The BIGTREETECH documentation for the EBB SB2209 CAN RP2040 pictures the following. (I **realize** the new menuconfig is now Katapult and not CanBoot)

[image]

Arksine, 2023-12-04:
The first screenshot is the default initial configuration.  You have to select the RP2040 within the `Micro-controller Architecture` sub-menu.

TimGehres, 2023-12-04:
Hi Eric,

Thanks for getting back with me and pointing me in the right direction. I thought it would be something simple. I appreciate your help

Thanks again,
Tim

Arksine, 2023-12-04:
You're welcome.  I'll go ahead and close this, hope that the build/upload goes smoothly.

closed 2023-12-04 by Arksine