Arksine/katapult #106: flash_can.py not going into boot loader mode

treaves, 2024-01-22:
I'm trying to upgrade Klipper on a Manta M8P board that was already running successfully.  The latest version of Klipper requires the MCU's be updated as well.

When I run the 
`python3 ~/katapult/scripts/flash_can.py -r -u your-uuid`
command, it does not report an error and the 
`Bus 001 Device 005: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter`
device disappears.  But nothing ever shows up in _/dev/serial/by-id/_.

Has anyone overcome this issue?

I've also tried to manually go into DFU mode by pressing the boot0 & reset pins, but, that just seems to lock up the CB4 and I have to power-cycle.

Bazeone, 2024-09-09:
I'm having the same issue with an Octopus STM32F446. It was working fine 2 version ago. came problem with flashtool.py and flash_can.py.

Bazeone, 2024-09-10:
working on v0.0.1-75-g90eb71b6.

Igor-Yavych, 2024-12-30:
I still have the same issue. OpenMoko device disappears from the `lsusb` list, but does not enter DFU mode. It ends up in some kind of limbo. 
**Board:** Manta M8P v2
**Katapult:** Latest
**Klipper:** Latest

Igor-Yavych, 2025-01-06:
Configuring Katapult to use USB on PA11/PA12 for Manta M8P v2 and then requesting bootloader via flashtool will somewhat work (unlike when it's set to canbus) and will allow Manta to be flashed via serial, but not via DFU