Arksine/katapult #109: Supported STM32 models

T3AR5, 2024-01-29:
Hi and first of all thank you for your work on CAN integration for klipper!

I am looking to design my own CAN toolheadboard to use in an insulated chamber where most of the current available toolheadboards seem to struggle due to only max. 85°C rated components.

Before I start I would like to get your help on selecting a suitable MCU as I am relatively new to the software aspect while the electronics should not be a problem. 

The MCU I currently favor is the STM32G473CCT3, but you have only the F series listed as supported. 

The only F series one I could find that is rated for atleast 125°C was STM32F413CHU3 and twice the price. 

Your help on this would be appreciated :)
Best regards

JNP-1, 2024-05-23:
Have you also looked at the STM32G0B1 series?
The STM32G0B1CBT3 goes up to 125°C and is supported by katapult.

T3AR5, 2024-05-23:
hey,
thanks for your reply! I actually decided to use the one you suggest, but in a UFQFPN package, so the STM32G0B1CBU3.
Apart from that I didnt have to much time and just recently actualy started with the schematics :'D