Arksine/katapult #111: [Help needed] - cause of "CONNECT timed out"on btt-mmb

lesaux, 2024-01-31:
Hello, I have two STM32G0B1 devices, one is the BTT ercf board (MMB) and the other is an EBB42 toolboard.

I can scan for both devices with the flash_can.py tool.

```
Resetting all bootloader node IDs...
Checking for Katapult nodes...
Detected UUID: 201615ff90d9, Application: Klipper. <--manta-m8p
Detected UUID: 2b12809532d4, Application: Klipper <--ebb42
Detected UUID: d04f5a660b7f, Application: Klipper <--mmb
Query Complete
```

When I try to flash the mmb, it times out.

```
~/klippy-env/bin/python ~/CanBoot/scripts/flash_can.py -u d04f5a660b7f -f ../klipper_flasher/klipper-mmb.bin
Sending bootloader jump command...
Resetting all bootloader node IDs...
Attempting to connect to bootloader
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/biqu/CanBoot/scripts/flash_can.py", line 475, in run
    await flasher.connect_btl()
  File "/home/biqu/CanBoot/scripts/flash_can.py", line 89, in connect_btl
    ret = await self.send_command('CONNECT')
  File "/home/biqu/CanBoot/scripts/flash_can.py", line 193, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [CONNECT] to Can Device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/biqu/CanBoot/scripts/flash_can.py", line 622, in main
    loop.run_until_complete(sock.run(intf, uuid, fpath, req_only))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/biqu/CanBoot/scripts/flash_can.py", line 483, in run
    await flasher.finish()
  File "/home/biqu/CanBoot/scripts/flash_can.py", line 272, in finish
    await self.send_command("COMPLETE")
  File "/home/biqu/CanBoot/scripts/flash_can.py", line 193, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [COMPLETE] to Can Device
```

If I query the canbus again, I find the MMB, but this time in bootloader mode:

```
Detected UUID: d04f5a660b7f, Application: Katapult
```

Same method works fine with the ebb42.
What could be wrong here?

Riley903, 2024-02-07:
I'm having the exact same issue but with the EBB42. Have you just been following the guide in Arksine's README? What differences did you make for the EBB42?

I'm currently using a BTT U2C V2.1 to give me the bridge which is connected to a Pi4.

lesaux, 2024-02-07:
I think my issue is more of how I setup 3 devices on canbus. In klipper (devices that have 2 wires) 3 devices would just be in a "star". You can't really have a daisy chain. How to pick which devices that need terminators? Wire length?
In RRF (with 4 wires) the daisy chain makes more sense. There's something I don't understand with klipper canbus and more than two devices. I need to play with it more.

Riley903, 2024-02-08:
It's true, having a "star" configuration is not the most ideal. I believe you are correct assuming wire length is the deciding factor for where you put the terminations. I'm also running an M8P with an EBB42, I don't want to have the M8P as my bridge since I'm planning on having multiple CAN networks (one for high priority requirements such as the motion system, and other networks for low demand systems), I currently terminate at the EBB42 and the M8P since they both split off from the U2C (U2C in the "middle" of the CAN transmission lines). I want to be able to flash the CAN clients/nodes (in my case the M8P and EBB42) over the CAN network, long term goal is to be able to flash CAN devices without any manual intervention (no BOOT button pressing or using something like STM32CubeProgrammer every time I want to update the firmware). Its seems like katapult/CanBoot is theoretically capable of this, but I'm struggling to get past the issue you are experiencing as well. 

How were you able to flash the EBB42 over the CAN network?

lesaux, 2024-02-08:
Oh, I can flash the EBB42 or the MMB reliably if I only have two devices on the bus, the main MCU and either the ebb or the mmb.
My purpose was the same as yours, be able to flash all 3 devices easily without manual intervention.
I also experimented quite a bit with the duet distribution toolboard, but I think I figured out it just can't daisy chain devices properly with two wires - in this case it's just doing a "Y". I tried moving resistors around in all possible combinations on the three devices to no effect. I'm a bit stumped!
I've ordered a cheap mellow can bus expander, but from the little I could find about it online, it seems very basic, and I have a feeling it's not going to help.

Riley903, 2024-02-09:
Hey, I managed to get mine to work. What fixed it for me was actually using the flash_can.py flashing tool in the klipper/lib/canboot rather than using katapult's. I guess using previously working software is more reliable than the latest version? I'd recommend trying that to see if it works for you, otherwise let me know how I could recreate your issue and see if I can get past the:

```
Sending bootloader jump command...
Resetting all bootloader node IDs...
Attempting to connect to bootloader
INFO:root:Response for command CONNECT timed out, 4 tries remaining
INFO:root:Response for command CONNECT timed out, 3 tries remaining
INFO:root:Response for command CONNECT timed out, 2 tries remaining
INFO:root:Response for command CONNECT timed out, 1 tries remaining
INFO:root:Response for command CONNECT timed out, 0 tries remaining
INFO:root:Response for command COMPLETE timed out, 4 tries remaining
INFO:root:Response for command COMPLETE timed out, 3 tries remaining
INFO:root:Response for command COMPLETE timed out, 2 tries remaining
INFO:root:Response for command COMPLETE timed out, 1 tries remaining
INFO:root:Response for command COMPLETE timed out, 0 tries remaining
ERROR:root:Flash Error
``` 

I'm going to try flashing multiple devices/nodes on multiple networks (a handful of EBB42 V1.2 and M8P V1.1 boards for now) to see how robust the process is.

lesaux, 2024-02-09:
Wow, thank you for that!
my script can now flash all boards in a row!

https://asciinema.org/a/4RzLf68IcEUeRTgGaiGUX6uaA

I guess I can close the issue

closed 2024-02-09 by lesaux

Arksine, 2024-02-09:
Interesting. The scripts haven't diverged much, but one difference is the script in the katapult repo removed the UUID query before flashing.  This was done because performing a query when multiple nodes are on the network can result in arbitration errors and cause nodes to fall off the network (it takes a lot of nodes or a lot of consecutive UUID queries to do this).

When the UUID query was removed it also removed a delay between node reset and node assignment.  I suspect this is what is causing the issue, and will push a fix to the script in this repo.

Creo2005, 2024-02-09:
> Wow, thank you for that! my script can now flash all boards in a row!
> 
> https://asciinema.org/a/4RzLf68IcEUeRTgGaiGUX6uaA
> 
> I guess I can close the issue
Please share the text of the script itself for updating.

lesaux, 2024-02-09:
> > Wow, thank you for that! my script can now flash all boards in a row!
> > https://asciinema.org/a/4RzLf68IcEUeRTgGaiGUX6uaA
> > I guess I can close the issue
> 
> Please share the text of the script itself for updating.

Oh this feature will be coming "for real" in klipper sometime 2024. My bash script is quite crummy, but I'll see if I can publish it to github over the weekend.

Creo2005, 2024-02-09:
> Oh this feature will be coming "for real" in klipper sometime 2024. My bash script is quite crummy, but I'll see if I can publish it to github over the weekend.

I made own script for updating the srk3 and ebb42 klipper  by katapult. I wanted to see how you did it, maybe I'll borrow something interesting :)

lesaux, 2024-02-09:
> I made own script for updating the srk3 and ebb42 klipper by katapult. I wanted to see how you did it, maybe I'll borrow something interesting :)

Here you go!
My method of "freeing up" the can bus devices from klipper for flashing might not be the greatest and I'm saving the kernel configs in a folder. Those configs wouldn't change very often, but it might improve the script to make the baud rate selectable.
Aside from that. this is just looping the same bash commands we usually run.
https://github.com/lesaux/klipper_canbus_flasher

Creo2005, 2024-02-09:
> Here you go! My method of "freeing up" the can bus devices from klipper for flashing might not be the greatest and I'm saving the kernel configs in a folder. Those configs wouldn't change very [...]

Seems very complicated to me, I made it much easier. and everything works. I will post it here in the morning

Creo2005, 2024-02-10:
> Here you go! My method of "freeing up" the can bus devices from klipper for flashing might not be the greatest and I'm saving the kernel configs in a folder. Those configs wouldn't change very [...]

My work script
For BTT SRR3
#!/bin/bash
#chmod u+x /home/creo/printer_data/config/script/update_skr3.sh

sudo service klipper stop
cd ~/klipper

# Update mcu skr3
echo "Start update mcu SKR3"
echo ""
make clean
make -j4 KCONFIG_CONFIG=/home/creo/printer_data/config/script/config.klipper.skr3
read -p "mcu SKR3 firmware built, please check above for any errors. Press [Enter] to continue flashing, or [Ctrl+C] to abort"
python3 ~/katapult/scripts/flash_can.py -d  /dev/serial/by-id/usb-katapult_stm32h723xx_410016000651313332323730-if00
read -p "mcu SKR3 firmware flashed, please check above for any errors. Press [Enter] to continue, or [Ctrl+C] to abort"
echo "Finish update mcu SKR3"
echo ""

sudo service klipper start

Arksine, 2024-02-10:
I pushed the change to the script in this repo.  It would be great if one of you could try it out and let me know if it resolved the issue.  Thanks.

lesaux, 2024-02-11:
I just tested the new version and it indeed worked on all 3 of my devices.
Thank you!