Arksine/katapult #114: EBB32 1.2 cannot flash klipper or canboot update via CAN

sredoje1973, 2024-03-12:
Hi there,
I'm trying to get klipper to flash on the BTT EBB 42 1.2 via using canboot over canbus. I keep getting a "Error sending command [CONNECT] to Can Device" error when trying to flash.  The "Attempting to connect to bootloader" command takes a few seconds.   All details below.  Any idea what I might be doing wrong? 

I am using a BTT PAD 7 with CM4 installed.  The CAN configuration is selected correctly (have completed a resonance test with the supplied ADXL345 over SPI. 

I have flashed canboot following the instructions here. https://docs.meteyou.wtf/btt-ebb/canboot/.  That seemed to work successfully (blinking blue led).

After flashing I disconnected the USB from the pi (moved it to a brick), connected the CAN interface (H to H, L to L), jumpered the 120ohm jumper. When plugged it all in. 

When I query the CAN network, I get a UUID per below.  That tells me that CAN communication is working.
Tried to flash klipper several times over CAN but got the same errors below.  I thought I'd try to just update canboot over CAN (in case I was doing something very wrong)

-----------------------------------------------------
Setting from my pi:
can0: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1024  (UNSPEC)
        RX packets 39  bytes 264 (264.0 B)
        RX errors 8  dropped 0  overruns 0  frame 8
        TX packets 24  bytes 134 (134.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

biqu@btt-pad7:~/katapult/scripts $ cat /etc/network/interfaces.d/can0
allow-hotplug can0
iface can0 can static
   bitrate 1000000
   up ifconfig $IFACE txqueuelen 1024
---------------------------------------------------------------------
 my menuconfig options
--------------------------------------------------------------------
                                                                                Katapult Configuration v0.0.1-64-g3e23332
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32G0B1)  --->
    Build Katapult deployment application (8KiB bootloader)  --->
    Clock Reference (8 MHz crystal)  --->
    Communication interface (CAN bus (on PB0/PB1))  --->
    Application start offset (8KiB offset)  --->
(1000000) CAN bus speed
()  GPIO pins to set on bootloader entry
[*] Support bootloader entry on rapid double click of reset button
[ ] Enable bootloader entry on button (or gpio) state
[*] Enable Status LED
(PA13)  Status LED GPIO Pin

-------------------------------------------------------------------------
Commands and error messages below:
--------------------------

biqu@btt-pad7:~/katapult/scripts $ python3 flashtool.py -i can0 -q
Resetting all bootloader node IDs...
Checking for Katapult nodes...
Detected UUID: 4220d6e9e9f9, Application: Katapult
Query Complete
biqu@btt-pad7:~/katapult/scripts $ python3 ~/katapult/scripts/flash_can.py -f ~/katapult/out/canboot.bin -i can0 -u 4220d6e9e9f9
Sending bootloader jump command...
Resetting all bootloader node IDs...
Attempting to connect to bootloader
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/biqu/katapult/scripts/flash_can.py", line 626, in main
    loop.run_until_complete(sock.run(intf, uuid, fpath, req_only))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/biqu/katapult/scripts/flash_can.py", line 479, in run
    await flasher.connect_btl()
  File "/home/biqu/katapult/scripts/flash_can.py", line 90, in connect_btl
    ret = await self.send_command('CONNECT')
  File "/home/biqu/katapult/scripts/flash_can.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
**FlashCanError: Error sending command [CONNECT] to Can Device**

NAPCAL, 2024-03-12:
If you already have the UUID set in your configuration file then you need to stop the Klipper service so it will release its hold on the CAN bus device.

Also, if you used the default Katapult firmware settings you can double click the reset to get into firmware loading mode to see if Katapult is still running on that device.

sredoje1973, 2024-03-12:
Thanks for the speedy reply.

1. Have not made any changes to klipper config (so yes, I deviated from the instructions a bit.... :) .  I wanted to get all the firmware working first before trying to get my pi to talk to the firmware.
2. I used the default settings in Katapult, but if I double klick on the reset button, what would happen? And does that tell me if katapult is working properly?

In parallel, I'm going to try to flash klipper via USB and see if that works. If so, then it narrows down the problem to something likely related to katapult and my setup

sredoje1973, 2024-03-12:
Below I used the verbose options and also include the dump files from my attempts to run those commands

biqu@btt-pad7:~/katapult/scripts $ python3 flashtool.py -i can0 -q -v
Resetting all bootloader node IDs...
Checking for Katapult nodes...
Detected UUID: 4220d6e9e9f9, Application: Katapult
Query Complete
biqu@btt-pad7:~/katapult/scripts $ python3 flashtool.py -i can0 -b 1000000 -v -f ~/klipper/out/klipper.bin -u 4220d6e9e9f9
Sending bootloader jump command...
Resetting all bootloader node IDs...
Attempting to connect to bootloader
INFO:root:Response for command CONNECT timed out, 4 tries remaining
INFO:root:Response for command CONNECT timed out, 3 tries remaining
INFO:root:Response for command CONNECT timed out, 2 tries remaining
INFO:root:Response for command CONNECT timed out, 1 tries remaining
INFO:root:Response for command CONNECT timed out, 0 tries remaining
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/biqu/katapult/scripts/flashtool.py", line 626, in main
    loop.run_until_complete(sock.run(intf, uuid, fpath, req_only))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/biqu/katapult/scripts/flashtool.py", line 479, in run
    await flasher.connect_btl()
  File "/home/biqu/katapult/scripts/flashtool.py", line 90, in connect_btl
    ret = await self.send_command('CONNECT')
  File "/home/biqu/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [CONNECT] to Can Device

dump below
;--------------------------------------------------------

biqu@btt-pad7:~ $ candump -tA -Ddex can0,#FFFFFFFF >> mycanlog
^Cbiqu@btt-pad7:~ cat mycanlog
 (2024-03-12 16:38:39.399732)  can0  TX - -  3F0   [1]  12
 (2024-03-12 16:38:39.900924)  can0  TX - -  3F0   [1]  00
 (2024-03-12 16:38:39.901145)  can0  RX - -  3F1   [8]  20 42 20 D6 E9 E9 F9 11
 (2024-03-12 16:38:50.800444)  can0  TX - -  3F0   [7]  02 42 20 D6 E9 E9 F9
 (2024-03-12 16:38:51.301511)  can0  TX - -  3F0   [1]  12
 (2024-03-12 16:38:52.303163)  can0  TX - -  3F0   [8]  11 42 20 D6 E9 E9 F9 81
 (2024-03-12 16:38:52.804536)  can0  TX - -  202   [8]  01 88 11 00 F1 7C 99 03
 (2024-03-12 16:38:52.804758)  can0  RX - -  203   [8]  01 88 A0 07 11 00 00 00
 (2024-03-12 16:38:52.804971)  can0  RX - -  203   [8]  00 00 01 00 00 20 00 08
 (2024-03-12 16:38:52.805178)  can0  RX - -  203   [8]  40 00 00 00 73 74 6D 33
 (2024-03-12 16:38:52.805337)  can0  RX - -  203   [4]  1B 76 99 03
 (2024-03-12 16:38:52.805451)  can0  RX - -  20000004   [8]  00 09 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-overflow,tx-error-warning}
 (2024-03-12 16:38:55.010011)  can0  TX - -  202   [8]  01 88 11 00 F1 7C 99 03
 (2024-03-12 16:38:55.010231)  can0  RX - -  203   [8]  01 88 A0 07 11 00 00 00
 (2024-03-12 16:38:55.010434)  can0  RX - -  203   [8]  00 00 01 00 00 20 00 08
 (2024-03-12 16:38:55.010635)  can0  RX - -  203   [8]  40 00 00 00 73 74 6D 33
 (2024-03-12 16:38:55.010837)  can0  RX - -  203   [4]  1B 76 99 03
 (2024-03-12 16:38:55.010946)  can0  RX - -  20000004   [8]  00 09 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-overflow,tx-error-warning}
 (2024-03-12 16:38:57.217541)  can0  TX - -  202   [8]  01 88 11 00 F1 7C 99 03
 (2024-03-12 16:38:57.218077)  can0  RX - -  203   [8]  01 88 A0 07 11 00 00 00
 (2024-03-12 16:38:57.218489)  can0  RX - -  203   [8]  00 00 01 00 00 20 00 08
 (2024-03-12 16:38:57.218784)  can0  RX - -  20000004   [8]  00 09 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-overflow,tx-error-warning}
 (2024-03-12 16:38:57.219302)  can0  RX - -  203   [4]  1B 76 99 03
 (2024-03-12 16:38:59.426536)  can0  TX - -  202   [8]  01 88 11 00 F1 7C 99 03
 (2024-03-12 16:38:59.427073)  can0  RX - -  203   [8]  01 88 A0 07 11 00 00 00
 (2024-03-12 16:38:59.427479)  can0  RX - -  203   [8]  00 00 01 00 00 20 00 08
 (2024-03-12 16:38:59.427768)  can0  RX - -  20000004   [8]  00 09 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-overflow,tx-error-warning}
 (2024-03-12 16:38:59.428281)  can0  RX - -  203   [4]  1B 76 99 03
 (2024-03-12 16:39:01.635476)  can0  TX - -  202   [8]  01 88 11 00 F1 7C 99 03
 (2024-03-12 16:39:01.636015)  can0  RX - -  203   [8]  01 88 A0 07 11 00 00 00
 (2024-03-12 16:39:01.636424)  can0  RX - -  203   [8]  00 00 01 00 00 20 00 08
 (2024-03-12 16:39:01.636715)  can0  RX - -  20000004   [8]  00 09 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-overflow,tx-error-warning}
 (2024-03-12 16:39:01.637231)  can0  RX - -  203   [4]  1B 76 99 03
 (2024-03-12 16:39:03.844781)  can0  TX - -  202   [8]  01 88 15 00 91 1B 99 03
 (2024-03-12 16:39:03.845332)  can0  RX - -  203   [8]  01 88 A0 01 15 00 00 00
 (2024-03-12 16:39:03.845753)  can0  RX - -  203   [4]  00 2E 99 03

Townie-au, 2024-04-17:
I also am having this issue.
Katapult seems to flash but doesnt pick up katapult node in ssh to flash klipper.