Arksine/katapult #115: Can't get this to work on Fly ERCF or BTT Manta M8P

alidaf, 2024-03-14:
I've been struggling with this since I bought these nearly a year ago, when Katapult was still Canboot. I'm having another go but getting nowhere with either of these boards. I seem to be able to get the bootloader onto the M8P but then nothing else beyond that.
The ERCF should be much simpler and has step by step instructions [here](https://mellow-3d.github.io/fly_ercf_v1.1_canboot_usb.html). I've tried both CanBUS and USB but both fail. USB fails to provide a serial by-id and CanBUS fails to give me a CAN ID.
The M8P instructions are all over the place, often conflicting, and in many places out of date but there remains no succinct guide for Katapult that I've managed to get working. The best there is is [here](https://github.com/Arksine/katapult/issues/61), but it veers off course, becomes confusing and out of date in parts.
 I am also using a CB1 with Armbian and this may be the root of the issue but this seems to be the unofficially supported distro for BTT instead of their out of date version of Raspbian.

Creo2005, 2024-03-14:
try by analogy only the config of your board

Dersonje, 2024-04-28:
Same here. Followed other pages but flashing in DFU mode doesn't work either. I see the CAN adapter but no /dev/serial is made

alidaf, 2024-04-29:
> Same here. Followed other pages but flashing in DFU mode doesn't work either. I see the CAN adapter but no /dev/serial is made

Try [this](https://canbus.esoterical.online/). It helped me get everything working. The problem is that there are so many posts all over the place with differing and even conflicting instructions but this guide has been pretty much definitive for me.

closed 2024-04-29 by alidaf