Arksine/katapult #120: Is posible reduce size of the final binary ?

semos2k, 2024-04-27:
Hi, i used katapult on multiples MCU and work great, but i tried on stm32f042 this contains 32kb flash. I tried to compile katapult using application start offset 4k but final binary is more large than this size. My intention is install klipper but the size of the binary and katapult is more than 32k.

Is posible reduce this binary ? reduce the number off features ?

WegnerDan, 2024-06-01:
I have the same issue. Trying to build katapult for the voron 0 display and expander MCUs.

The "Initial Release" has some compiled binaries that are below 4096 byte, but I'm wondering if this is still doable

[image]

<details>
<summary>make output</summary>

```
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/bootentry.o
  Compiling out/src/command.o
  Compiling out/src/flashcmd.o
  Compiling out/src/generic/armcm_canboot.o
  Compiling out/src/initial_pins.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/flash.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/../lib/stm32f0/system_stm32f0xx.o
  Compiling out/src/stm32/stm32f0.o
  Compiling out/src/stm32/stm32f0_timer.o
  Compiling out/src/stm32/gpioperiph.o
  Compiling out/src/stm32/usbfs.o
  Compiling out/src/stm32/chipid.o
  Compiling out/src/generic/usb_cdc.o
  Preprocessing out/src/generic/armcm_link.ld
  Compiling out/src/deployer.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_reset.o
  Preprocessing out/src/generic/armcm_deployer.ld
  Building out/compile_time_request.o
  Linking out/katapult.elf
  Building out/deployer_ctr.o
  Creating bin file out/katapult.bin

The Katapult binary is too large for the configured LAUNCH_APP_ADDRESS.

Rerun "make menuconfig" and either increase the LAUNCH_APP_ADDRESS or
disable features to reduce the final binary size.

Maximum size 4096. Current size 4328.

make: *** [Makefile:85: out/katapult.bin] Error 255
```

</details>

WegnerDan, 2024-06-01:
I was able to flash the old canboot binary from "Initial Release" and use it to update klipper on the v0 display and klipper expander MCUs. Now the boot jumper isn't necessary anymore.

janderholm, 2024-09-11:
@WegnerDan the configuration you posted comes out to 3928 bytes using master of today with compiler version:

arm-none-eabi-gcc (Arm GNU Toolchain 13.3.Rel1 (Build arm-13.24)) 13.3.1 20240614

That compiler version was released in 14th of June after your comment, so perhaps you may want to revisit it to use a new version of katapult?

(I haven't actually tested the compiled binary)