Arksine/katapult #121: Flashing RP2040 - "No rule to make target 'flash'. Stop.

Burnsian, 2024-05-18:
I read another issue thread with this exact issue, so perhaps I should have been smart enough to understand your answer and get what I needed. Sadly, I'm too much of a noob and need some more hand holding to understand what to do next. 

I'm following BigTreeTech's manual on the SB2209 CAN (RP2040), page 29. After setting the SB2209 into 'boot' mode, querying DFU device ID, I came up with 2e8a:0003, identical to the ID shown in the guide. Then I entered the next command in the manual verbatim, "make flash FLASH_DEVICE=2e8a:0003", then I get the same error code "no rule to make target 'flash'". I tried following your advice, Arksine, and just typing "make flash" here, but that didn't work either. 
Any help would be appreciated!

_Originally posted by @Burnsian in https://github.com/Arksine/katapult/issues/96#issuecomment-2118836517_

Burnsian, 2024-05-23:
Found a tutorial elsewhere

closed 2024-05-23 by Burnsian

Dmandjob, 2024-07-26:
What was the solution?