Arksine/katapult #123: Custom Board w/ STM32G0B1CEU6 - 512K

syph3rd, 2024-05-30:
Hey Guys,

I have a custom board using an STM32G0B1CEU6, I have tried to flash Katapult viaan ST Link but the board refuses to operate and i need to hold it to ground while connecting in via SWD to get the chip in a useable state.

being that the chip is 512K vs 128K would this cause issues?

Thanks

Arksine, 2024-05-30:
I don't think the amount of onboard flash should matter.  As a general rule Katapult runs a subset of MCUs supported by Klipper....so one can always try to see if it will run Klipper first, then see if Katapult is supported.

If Klipper runs successfully but Katapult does not then the only other advice I can provide for a custom design is to use a debugger to see what is going on when the application is loaded.

syph3rd, 2024-05-30:
Thanks Eric,
I spent some time with it and have it functional.

Thank you for your response.