Arksine/katapult #125: Problem using Katapult with a serial connection

mykepredko, 2024-08-01:
I don't know if you've seen this post on the Klipper Discourse group:

[Trying to use Katapult with a serial connection](https://klipper.discourse.group/t/trying-to-use-katapult-with-a-serial-connection/18012)

I've loaded up Katapult on my SKR Mini E3 V3.1 with a serial connection from an Orange Pi Zero 2W but when I try to load Klipper using flashtool.py, I get a strange error that's listed in the link above.  

Can you comment?  I suspect I'm doing something wrong but I can't find any reference to doing this.  

Thanx!

lktop7, 2024-08-13:
It seems way too simple but could it be that you selected the wrong processor. Afaik the board exists in two versions, the v3 with the stm32g0b1 and the version v3.0.1 stm32f401. Which version are you using?

lktop7, 2024-08-14:
After finding an old STM 32 demo board with an integrated ST-LINK programmer in my magic box, I mustered up the courage to compile and flash Catapult with UART for the STM32G0 version of the board. It worked despite a small pitfall, the klipper host process interfered with the flashing of klipper on the MCU.

`cd ../../katapult/
make menuconfig`
[image]
`make
./scripts/flashtool.py -d /dev/serial/by-id/usb-katapult_stm32g0b1xx_0C004A0014504B5735313920-if00 -v -f out/deployer.bin`
-> flash uart deployer with the previous USB Katapult
`/home/[my_username]/katapult/./scripts/flashtool.py:619: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
Flashing Serial Device /dev/serial/by-id/usb-katapult_stm32g0b1xx_0C004A0014504B5735313920-if00, baud 250000
Attempting to connect to bootloader
INFO:root:Response for command CONNECT timed out, 4 tries remaining
Katapult Connected
Software Version: v0.0.1-69-g562fc2f
Protocol Version: 1.1.0
Block Size: 64 bytes
Application Start: 0x8002000
MCU type: stm32g0b1xx
Flashing '/home/[my_username]/katapult/out/deployer.bin'...

[INFO:root:Command 'SEND_BLOCK': Acknowledged wrong command, expected: 12, received: 11
##################################################]

Write complete: 3 pages
Verifying (block count = 70)...

[#################################################]

Verification Complete: SHA = 432485BAB7DB5514C8D38E03C8E6B8697255CB7E
Flash Success

../katapult/scripts/flashtool.py -d /dev/ttyS1 -v -f out/klipper.bin 
/home/[my_username]/klipper/../katapult/scripts/flashtool.py:619: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
Flashing Serial Device /dev/ttyS1, baud 250000
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 385, in _reconfigure_port
    fcntl.flock(self.fd, fcntl.LOCK_EX | fcntl.LOCK_NB)
BlockingIOError: [Errno 11] Resource temporarily unavailable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/[my_username]/klipper/../katapult/scripts/flashtool.py", line 555, in run
    serial_dev.open()
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 387, in _reconfigure_port
    raise SerialException(msg.errno, "Could not exclusively lock port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 11] Could not exclusively lock port /dev/ttyS1: [Errno 11] Resource temporarily unavailable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/[my_username]/klipper/../katapult/scripts/flashtool.py", line 643, in main
    loop.run_until_complete(sock.run(args.device, args.baud, fpath))
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/[my_username]/klipper/../katapult/scripts/flashtool.py", line 557, in run
    raise FlashError("Unable to open serial port: %s" % (e,))
FlashError: Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/ttyS1: [Errno 11] Resource temporarily unavailable`
-> First try to flash klipper over UART failed
`sudo systemctl stop klipper

./katapult/scripts/flashtool.py -d /dev/ttyS1 -v -f out/klipper.bin 
/home/[my_username]/klipper/../katapult/scripts/flashtool.py:619: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
Flashing Serial Device /dev/ttyS1, baud 250000
Attempting to connect to bootloader
Katapult Connected
Software Version: v0.0.1-69-g562fc2f
Protocol Version: 1.1.0
Block Size: 64 bytes
Application Start: 0x8002000
MCU type: stm32g0b1xx
Flashing '/home/[my_username]/klipper/out/klipper.bin'...

[##################################################]

Write complete: 18 pages
Verifying (block count = 550)...

[##################################################]

Verification Complete: SHA = DF82346A46344D25670267236E6BECB129335FB0
Flash Success`
-> Second try to flash klipper over UART , after stopping the klipper process on the host, was successful

Arksine, 2024-09-03:
I'm going to close this as it has been resolved.

closed 2024-09-03 by Arksine