Arksine/katapult #129: Flashtool reboot (`-r`) fails with write error

ressu, 2024-09-09:
The latest version of flashtool.py seems to error out when trying to write to canbus. The error is visible when enabling verbose mode:

```
$ ~/katapult/scripts/flashtool.py -i can0 -u deadbeefbeef -r -v
Flashing CAN UUID deadbeefbeef on interface can0
Sending bootloader jump command...
Bootloader request command sent
Flash Success
INFO:root:Socket Write Error, closing
```

This is an issue introduced by the refactoring in 730fde48ab86ecb114ed1d6d930b53ba20e251bf.

The issue seems to trigger when creating a new task in the `send()` method in `CanSocket`. If I add a bit of additional logging, we can see the following:
```
$ ~/katapult/scripts/flashtool.py -i can0 -u deadbeefbeef -r -v
Flashing CAN UUID deadbeefbeef on interface can0
INFO:root:Socket in _jump_to_bootloader: <socket.socket fd=6, family=29, type=3, proto=1, laddr=('can0',)>
Sending bootloader jump command...
INFO:root:Socket in send: <socket.socket fd=6, family=29, type=3, proto=1, laddr=('can0',)>
Bootloader request command sent
Flash Success
INFO:root:Socket in _do_can_send: <socket.socket [closed] fd=-1, family=29, type=3, proto=1>
ERROR:root:Socket Write Error, Closing
Traceback (most recent call last):
  File "/home/biqu/katapult/scripts/flashtool.py", line 414, in _do_can_send
    await self._loop.sock_sendall(self.cansock, packet)
  File "/home/biqu/.pyenv/versions/3.11.10/lib/python3.11/asyncio/selector_events.py", line 536, in sock_sendall
    n = sock.send(data)
        ^^^^^^^^^^^^^^^
OSError: [Errno 9] Bad file descriptor
INFO:root:Socket Write Error, closing
```

Forcing a creation and opening of the socket in `_do_can_send()` sends the packet as expected, so this seems to be an issue on how asyncio and the socket that is opened inside the `run()` method interact. 

I've reproduced this on 2 boards:
- BTT Pi V1.2 using Armbian 24.5.1 bookworm and  Python 3.11.2
- BTT CB2 using a biqu based image (Debian 11) and Python 3.9.2 & 3.11.10

Arksine, 2024-09-09:
Ah, I suspect that the event loop is closed before the command is sent.  Commit 90eb71b610669aaaab5ed3705c5b6361ac50dbb6 should fix this.

ressu, 2024-09-09:
Looks like that was it. I can now trigger bootloader on both machines without issues.

closed 2024-09-09 by ressu