Arksine/katapult #137: MIPS support for flash_can.py?

pellcorp, 2024-11-28:
So I am using a cartographer on a K1, and so far we haver never managed to figure out how to flash the cartographer on the K1.  I suspect something needs to change in serialposix.py but I am unsure where to start looking.

So I know that Creality actually patched serialposix.py for the K1 build root already, and I am wondering if what they changed might provide some guidance?

[image]

I would appreciate any suggestions you might have, because this is what I am getting at the moment and its now locked up:

```
root@K1-0753 /usr/data/klipper/scripts [#] /usr/share/klippy-env/bin/python /usr/data/katapult/scripts/flash_can.py -f /usr/data/cartographer-klipper/firmware/v2-v3/survey/5.0.0/Survey_Cartographer_K1_USB
_8kib_offset.bin -d $CATAPULT_DEV -v
Flashing Serial Device /dev/serial/by-id/usb-katapult_stm32f042x6_16000C000F43304253373820-if00, baud 250000
Attempting to connect to bootloader
INFO:root:Response for command CONNECT timed out, 4 tries remaining
INFO:root:Response for command CONNECT timed out, 3 tries remaining
INFO:root:Response for command CONNECT timed out, 2 tries remaining
INFO:root:Response for command CONNECT timed out, 1 tries remaining
INFO:root:Response for command CONNECT timed out, 0 tries remaining
INFO:root:Response for command COMPLETE timed out, 4 tries remaining
INFO:root:Response for command COMPLETE timed out, 3 tries remaining
INFO:root:Response for command COMPLETE timed out, 2 tries remaining
INFO:root:Response for command COMPLETE timed out, 1 tries remaining
```

pellcorp, 2024-11-28:
Oh and cancelling corrupted katapult boot loader and I had to use DFU to reflash.

pellcorp, 2024-11-28:
This is the commit creality back ported in a creality way 

https://github.com/pyserial/pyserial/commit/5ce1773fdce16a0b184363695a7d7d9483c4da0c

lastowl, 2026-10-27:
I tracked this down on a K1 Max (SimpleAF on Kalico, Cartographer V4 USB, stm32g431, Katapult `v0.0.1-104-gfcb2f84`). It isn't pyserial or MIPS baud handling. Creality's pyserial 3.4 sets 250000 correctly, and current flashtool fails the same way.

**Cause.** Creality's `/etc/udev/rules.d/60-serial.rules` runs `/usr/bin/cx_ai_crypto $devnode` against every new `ttyACM` device to look for the AI lidar. That tool writes 8 bytes, `af ff 02 00 01 55 fa fd`, to whatever just enumerated, including Katapult right after the bootloader request.

`command_find_block()` then livelocks:

1. It discards `af ff 02 00` and resyncs to the `0x01`.
2. The block `01 55 ...` is invalid.
3. The error path's `memchr()` starts at `buf[0]`, finds that same `0x01`, and pops 0 bytes.
4. The receive buffer never drains. Once it holds more than 64 bytes, `usb_cdc` stops reading the bulk OUT endpoint.

From then on the bootloader NAKs all data until power is removed: CONNECT times out, then writes block. EP0 keeps working. A USB reset doesn't help because the stuck bytes are in RAM.

**Evidence**
- Ran `cx_ai_crypto` against a pty to capture the exact bytes.
- Talked to the bootloader through raw usbfs, bypassing `cdc_acm`. Bulk OUT is ACKed for a few packets, then NAKs forever. EP0 control requests still work, and a COMPLETE command sent after a USB reset is ACKed but never acted on.
- Compiled the real `command_find_block()` and the `usb_cdc` receive loop on a PC and fed them the probe bytes, then prime, then 5× CONNECT. Result: 0 frames dispatched, with the buffer stuck at 68 bytes starting `01 55`. A clean start dispatches every frame.
- With `cx_ai_crypto` disabled for one boot, **unmodified** flashtool.py connected on the printer. I then flashed CartographerV4 6.2.0 USB Lite from the K1 itself, and verification passed.

**Workaround on K1 series, lasts until reboot:**
```
/etc/init.d/S55klipper_service stop
mount --bind /bin/true /usr/bin/cx_ai_crypto
# then request the bootloader and flash as usual
```
A permanent fix is to skip the lidar probe for Klipper/Katapult devices. I tested this on the same printer, and stock `flashtool.py -s` connects with it in place:
```
sed -i 's|^KERNEL=="ttyACM\[0-9\]\*", IMPORT{program}="/usr/bin/cx_ai_crypto|KERNEL=="ttyACM[0-9]*", ENV{ID_VENDOR_ID}!="1d50", IMPORT{program}="/usr/bin/cx_ai_crypto|' /etc/udev/rules.d/60-serial.rules
udevadm control --reload
```
I'm suggesting that to SimpleAF as well. A Creality firmware update overwrites that file, so the installer would need to re-apply it.

**Fix for the resync:** #195. The search now starts after the first byte of the rejected block. Existing probes keep their current bootloader until it's replaced, so the udev change is what helps K1 users now.

The same udev rule probes every `ttyACM` device, so other Katapult-based probes on K1s should hit this too. I haven't tested that.

pellcorp, 2026-10-28:
I am closing this, it was a udev issue all along

closed 2026-10-28 by pellcorp