Arksine/katapult #140: Make issue within pico-sdk

sallen1301, 2024-12-27:
Trying to build for a FYSETC-ERB v2.0 with the following:
[image]

When I go do the make, I get the following errors:
[image]

If I flash to the board, the board will not boot into Katapult. I can flash the board with Klipper without issues

shortyishere, 2024-12-28:
I have same issue. Using hermit crab v2 with rp2040. It seems to be downloading to wrong regions.

SinisterRj, 2025-01-04:
Same here with BTT SKR pico.

Lebensgefahr, 2025-01-05:
> Trying to build for a FYSETC-ERB v2.0 with the following: <img alt="Screenshot 2024-12-26 at 6 17 36 PM" width="721" src="https://private-user-images.githubusercontent.com/77299733/398817582-8080a1a2-b064-4ee4-b91e-0ee53cfaee79.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNDE2NzUsIm5iZiI6MTczNjA0MTM3NSwicGF0aCI6Ii83NzI5OTczMy8zOTg4MTc1ODItODA4MGExYTItYjA2NC00ZWU0LWI5MWUtMGVlNTNjZmFlZTc5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDAxNDI1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgwYWUzOGVlNjEwNTExOGM2OGY0MGE3MTNlMTJjMGMxNDQ3YzllMjk2NGMzZDQ5ODUyOTA2MmFmNjNhYTA5YTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.qxpGQKeUBrGqzc0gn2oRKMqsSqE2_Ws9O359sHRCS10">
> 
> When I go do the make, I get the following errors: <img alt="Screenshot 2024-12-26 at 6 18 03 PM" width="809" src="https://private-user-images.githubusercontent.com/77299733/398817625-4ad5f0f6-635f-4239-834f-08f859912902.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNDE2NzUsIm5iZiI6MTczNjA0MTM3NSwicGF0aCI6Ii83NzI5OTczMy8zOTg4MTc2MjUtNGFkNWYwZjYtNjM1Zi00MjM5LTgzNGYtMDhmODU5OTEyOTAyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDAxNDI1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ3OGQ0YzNlNmVhMjc0YmY4MjU1NDgzZjMxMmUwOTI5YjQ0ODZjNDdlN2NkY2QzZGEyYmE1OTFmYWY2MGYzODAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.rRdm0MLA9nVJscCRnW31urJuVDG5tEIuk11xtAsMOe4">
> 
> If I flash to the board, the board will not boot into Katapult. I can flash the board with standalone Klipper without issues

AFAIK it is not an errors, it's just a warnings you can remove them by changing