Arksine/katapult #149: Unable to reflash STM32 using Katapult after first time

sshantt, 2025-02-15:
I am trying to get Katapult working on my STM32H732 board. I want to get it working so that I can update firmware remotely. 

I first connected the board to my macbook and flashed Katapult. I set the LED pin correctly and as soon as the flash was over the LED started blinking once every two seconds indicating that the board was in Katapult flash mode. Still on my macbook, I proceeded to use the Katapult python commands in order to flash Klipper on there. I made sure the Klipper offset was set to 128kb when creating the firmware file. The operation was successful and I tested klipper on a raspberry pi that had mainsail and klipper installed and verified it worked perfectly. 

The next step was to verify Katapult could enter into Katapult flash mode remotely. So I connected it back to my Macbook (does not have klipper installed) and tried to flash again using the same Katapult python command. This resulted in failure with the following message "Attempting to connect to bootloader INFO:root:Response for command CONNECT timed out". Clearly Katapult was no longer able to put the board into katapult flash mode anymore for some reason. I tried to connect it to the raspberry pi and try to flash it remotely that way but got the same message. 

Since I enabled the "Support bootloader entry on rapid double click of reset button" option, I tried to do that in order to check if the board would go into katapult flash mode. This didnt work since the LED didnt start flashing like it did when I first installed katapult. 

Next I connected the board to the macbook using DFU mode to check if Katapult was still on the STM32 chip. Reading the content showed me that the first 6kb or so were identical to the katapult.bin file I used initially. Next there was 122kb of empty space followed by the 31kb of data which matched up with the klipper.bin I used. As far as I could tell, Katapult was still on there and hadnt been replaced. 

I tried all different baud rates, I tried stopping all processes that were using the particular USB interface and a few more things but was not able to get the python code to get past "Attempting to connect to bootloader". I tried to completely clear the flash on the STM32 chip using STM32Cubeprogrammer and installing Katapult again from scratch and then using it to install klipper, and got the same results. Not sure what to try next. Has anyone experienced this or knows what I should do?

AlatusAles, 2025-02-15:
Обновление Клиппер для вашего STM.

~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0    ### просмотр **ЮЮИД**
    затем
cd ~/katapult/scripts
    затем
python3 flashtool.py -u **ЮЮИД** -r -v                                          ### перевод в загрузчик по CAN
    затем 
cd ~
    затем
ls /dev/serial/by-id/*                                                                           ### просмотр **ЮСБ-ИДС**
затем 
cd ~/katapult/scripts 
затем 
python3 flashtool.py -d **ЮСБ-ИДС** -v                                          ### прошивка по сериал 
 
[image]

sshantt, 2025-02-15:
Thanks for the quick reply. The method you described is the method I've been using up until now. I have attached a picture to show. Perhaps there's a way to connect to Klipper using `python3 -m serial.tools.miniterm`? I could try to see what klipper replies to the commands directly?

[image]

AlatusAles, 2025-02-15:
Возможно, при создании файла прошивки Katapult Вы создаете "deployment"?

[image]

Три принтера с H723 от одного хоста (x64) по USB и, к сожалению, ни на одной из плат мне не удалось заставить работать"deployment". Предложенным выше способом обновляю Klipper без проблем.

sshantt, 2025-02-15:
[image]

sshantt, 2025-02-15:
How does this work? Does the python script give klipper instructions to reboot into the bootloader? Perhaps it's an issue with klipper not restarting the board. Since the initial flash using flashtool.py works just fine, I dont think there's something wrong with Katapult but rather klipper. Is this the part thats responsible?

```
/****************************************************************
 * Bootloader
 ****************************************************************/

// Handle reboot requests
void
bootloader_request(void)
{
    try_request_canboot();
    dfu_reboot();
}

```

AlatusAles, 2025-02-15:
Предположу, что Ваш 723 подключен по USB в режиме моста и прошит Klipper, тогда для перевода MCU в режим загрузчика (при этом MCU программно отсоединится от шины CAN) необходимо выполнить### flashtool.py -u <UUID> -r -v ### (где "-r" очень важно !!!). Но, меня смущает _"Since I enabled the "Support bootloader entry on rapid double click of reset button" option, I tried to do that in order to check if the board would go into katapult flash mode. This didnt work since the LED didnt start flashing like it did when I first installed katapult."_.

sshantt, 2025-02-15:
I am connected to the board directly through USB. There is no CAN involved at the moment. Should I still use ### flashtool.py -u UUID -r -v ###? I thought for USB I should be using ### python3 flashtool.py -d /dev/tty.usbmodem2101 -v -r ###

Regarding my statement below, what I meant was that I also tried to get the board to go into bootloader mode using rapid double click of the reset button but it didnt work.

"Since I enabled the "Support bootloader entry on rapid double click of reset button" option, I tried to do that in order to check if the board would go into katapult flash mode. This didnt work since the LED didnt start flashing like it did when I first installed katapult."

[image]

sshantt, 2025-02-15:
Here's my clipper firmware config if it helps

[image]

AlatusAles, 2025-02-15:
[image]
Именно по этому я использую режим моста USB-CAN даже на принтере с одним MCU... 
https://github.com/Arksine/katapult#request-bootloader-can-devices-only

AlatusAles, 2025-02-15:
PS: Для удобства вы можете не подключать MCU к MAC чтобы прошить через DFU.
Существует "dfu-util".

dfu-util -l    ### просмотр DFU_IDS
dfu-util -d DFU_IDS --alt 0 -s 0x08000000:mass-erase:force:leave -D ~/katapult/out/katapult.bin
или
dfu-util -d DFU_IDS --alt 0 -s 0x08000000:mass-erase:force:leave -D ~/klipper/out/klipper.bin

sshantt, 2025-02-16:
I figured out I've been doing "Support bootloader entry on rapid double click of reset button" wrong. I've been double pressing the "boot" button instead of the "reset" button 🤦🏼‍♂️ Now that I double press the reset button it has been successfully going into Katapult flash mode. 

But that doesnt explain why I haven't been able to get it to go into flash mode remotely *without* physically double pressing the reset button. I still think klipper is the problem as it is not correctly responding to the request from the Katapult python script

AlatusAles, 2025-02-17:
> Я понял, что неправильно делал "Поддержка входа в загрузчик при быстром двойном нажатии кнопки сброса". Я дважды нажимал кнопку "boot" вместо кнопки "reset" 🤦🏼‍♂️ Теперь, когда я дважды нажимаю кнопку сброса, он успешно переходит в режим прошивки Katapult.
> 
> Но это не объясняет, почему я не смог заставить его перейти в режим вспышки удаленно, _без_ физического двойного нажатия кнопки сброса. Я все еще думаю, что проблема в klipper, так как он некорректно отвечает на запрос от скрипта Katapult python

Как мне известно (я могу ошибаться), возможность удаленно перевести MCU в режим загрузчика существует только по шине CAN ("USB-CAN мост" тоже работает). Повторюсь... Даже на принтере с одним MCU я использую USB-CAN мост !!! , в другом случае мне бы пришлось дважды нажимать РЕСЕТ или выводить кнопку GPIO.

sshantt, 2025-02-17:
Ahh, interesting. I didn't pick that up from the documentation. Ok, if this is true then it's not a bug but rather a missing feature. Would it be terribly difficult to add that functionality? I know someone that can take a stab at it if its not too complicated

AlatusAles, 2025-02-17:
> Ahh, interesting. I didn't pick that up from the documentation. Ok, if this is true then it's not a bug but rather a missing feature. Would it be terribly difficult to add that functionality? I know someone that can take a stab at it if its not too complicated

[image]
RTFM ;) Удачи Вам!

Lebensgefahr, 2025-02-17:
> I figured out I've been doing "Support bootloader entry on rapid double click of reset button" wrong. I've been double pressing the "boot" button instead of the "reset" button 🤦🏼‍♂️ Now that I [...]

1. To put your mcu into katapult bootloader remotely klipper should be installed.
2. To put your mcu physical or virtual serial device you can try with firmware flashing tool updated from this PR https://github.com/Arksine/katapult/pull/142 (I added sending command to klipper for both virtual/physical serial devices). It is already tested for usb serial and serial devices by other users.

sshantt, 2025-02-17:
@Lebensgefahr It works!!! You beast! Thank you 🙏🏻

Lebensgefahr, 2025-02-18:
> [@Lebensgefahr](https://github.com/Lebensgefahr) It works!!! You beast! Thank you 🙏🏻

You are welcome and do not forget to close your issue ;-)

closed 2025-02-18 by sshantt