Arksine/katapult #150: STM32G0B1 using PB0/PB1 as canbus device does not show up using Katapult

jrlomas, 2025-02-22:
When the board if flashed with the latest katapult, doing:

```
python ~/klipper/scripts/canbus_query can0
```
shows nothing.

However if flashing the board with katapult, using commit 081918ad769d1f1104ca253a4a8ace02147c345d:
```
python ~/klipper/scripts/canbus_query can0
```

shows:
```
Found canbus_uuid=8de5271c1601, Application: CanBoot
```

Same build, same exact settings, just different commit.

Arksine, 2025-02-22:
I can't reproduce.  On a BTT EBB 42, which has the same MCU and CAN pins, I can query on the latest commit:

```
~/klipper/venv/bin/python ./scripts/canbus_query.py can0
Found canbus_uuid=73e494c07756, Application: CanBoot
Total 1 uuids found
```
With flashtool:
```
./scripts/flashtool.py -q
Resetting all bootloader node IDs...
Checking for Katapult nodes...
Detected UUID: 73e494c07756, Application: Katapult
Query Complete
```

It could be that your version of gcc is producing an invalid binary.  What is the output of:
```
arm-none-eabi-gcc --version
```

My testing was built on version 13.2.1.

jrlomas, 2025-02-22:
Same here... 

[image]

Not sure why one checkout (checkpoint) would produce an invalid binary vs a previous commit with the same exact tool.
It is also suspicious that the commit that works is before the rework related to ARM (RP2350) and the execute in ram code.

[image]

I can understand without a board you can replicate the issue, it is hard if not impossible to you to debug... 
Since I am able to easily replicate the problem, is there anything I can do on my end to provide more information to diagnose the problem?

Arksine, 2025-02-23:
The patch that executes code in RAM is exclusive to the RP2XXX series, it shouldnt impact STM.  There were a few changes made to inline some of the startup code which would affect all MCUs.  They shouldn't have an impact on the canbus though.

I suppose the first thing I would do is make sure that the board actually entered the bootloader on the newest commit.  If it has a status LED, configure it, as this is the best way to know when the bootloader is running without a debugger. 

If you don't have an LED I would recommend enabling the button option to enter the bootloader.  You can select a pin and pull it high (or low if you invert it) with a jumper wire.

jrlomas, 2025-02-23:
Hey Arksine, thank you.  I do have a debugger probe and I can connect it.  Let me try to set up the st-link v3 I have and find out if the device entered the bootloader, or where in the code it gets to.  I'll post here soon what I find. Thank you again very much for the advice on how to proceed.

jrlomas, 2025-04-22:
The boards now work fine and go into bootloader with the latest code.  Thank you for all of the work in March, seems to have fix it.

closed 2025-04-22 by jrlomas