Arksine/katapult #151: DFU mode after flashig stm32f401xc via flashtool.py in serial mode

IATkachenko, 2025-02-23:
Hello,

I'm trying to use Katapult for updating firmware at [Fysetc Cheetah v2.0](https://wiki.fysetc.com/Cheetah_Board_V20/).

For now I have: 
  - flashed Katapult (via `dfu-util -a 0 -d 0483:df11 --dfuse-address 0x08000000:mass-erase:force -D /home/igor/katapult/out/katapult.bin`)
  - flashed Klipper firmware (via `dfu-util -R -a 0 -d 0483:df11 -s 0x08008000:leave -D /home/igor/printer_data/firmware/klipper.bin.v0.12.0-439`)

All works fine and stable. 
Then I have tried to flash same Klipper firmware via `flashtool.py`:
```
cd /home/igor/klipper/scripts && \
python3 -c import flash_usb as u; u.enter_bootloader('/dev/serial/by-id/usb-Klipper_stm32f401xc_300034000751383531393138-if00') && \
cd /home/igor/katapult/scripts && \
python3 flashtool.py -d /dev/serial/by-id/usb-katapult_stm32f401xc_300034000751383531393138-if00 -f /home/igor/printer_data/firmware/
klipper.bin.v0.12.0-439 -v

Entering bootloader on /dev/serial/by-id/usb-Klipper_stm32f401xc_300034000751383531393138-if00

Flashing Serial Device /dev/serial/by-id/usb-katapult_stm32f401xc_300034000751383531393138-if00, baud 250000
Attempting to connect to bootloader
Katapult Connected
Software Version: v0.0.1-79-g25a23cd
Protocol Version: 1.1.0
Block Size: 64 bytes
Application Start: 0x8008000
MCU type: stm32f401xc
Flashing '/home/igor/printer_data/firmware/klipper.bin.v0.12.0-439'...

[##################################################]

Write complete: 2 pages
Verifying (block count = 502)...

[##################################################]

Verification Complete: SHA = 0E59DB04BCFFCDCAA7DEE33EE18846F6A6B2D62F
Flash Success
```
All looks fine, but board is falling to DFU mode:

```bash
# igor@voron:~/printer_data/firmware $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 034: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
```
(re-flashing by dfu-util works well)

Could you have a look what is going wrong with `flashtool.py`?

Arksine, 2025-02-23:
Neither flashtool nor Katapult have the capability to enter the system bootloader.  My guess is that Klipper is doing it, presumably because somehow its setting the USB boot flag when you perform the bootloader request.  Are you flashing pristine Klipper or a fork?

Try entering Katapult manually, via double reset or a configured button depending on your build config.

fex42, 2025-11-15:
I have the same issue now. I can flash Klipper via Katapult and after that it is in DFU mode. A cold start of the whole printer boots Klipper, but immediately after flashing, the Cheetah V2 is in bootloader mode and there seems to be no way to enter Klipper except a cold start.