Arksine/katapult #152: [Enhancement] flashtool: Support USB/CAN bridge flashing in one step

EmperorArthur, 2025-02-27:
Currently, a Klipper mcu acting in USB/CAN mode will successfully trigger rebooting into the bootloader, but will then fail to flash the device.

Two different commands must be run.  One to reset the device, and another to flash it via serial.  It would be convenient if Katapult could detect this situation and either warn the user, or flash the device automatically.

# Current Situation
## TwoCommand's Succeed
```
~/katapult/scripts/flashtool.py -u 154aab036e6a -q
~/katapult/scripts/flashtool.py -d /dev/serial/by-id/usb-katapult_stm32g0b1xx_3D0052001450425539383920-if00
```
## Single Command Fails
```
~/katapult/scripts/flashtool.py -u 154aab036e6a
Flashing CAN UUID 154aab036e6a on interface can0
Sending bootloader jump command...
Resetting all bootloader node IDs...
Attempting to connect to bootloader
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/pi/katapult/scripts/flashtool.py", line 499, in run
    await flasher.connect_btl()
  File "/home/pi/katapult/scripts/flashtool.py", line 99, in connect_btl
    ret = await self.send_command('CONNECT')
  File "/home/pi/katapult/scripts/flashtool.py", line 216, in send_command
    raise FlashError("Error sending command [%s] to Device" % (cmdname))
FlashError: Error sending command [CONNECT] to Device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/katapult/scripts/flashtool.py", line 661, in main
    await sock.run(intf, uuid, fpath, req_only)
  File "/home/pi/katapult/scripts/flashtool.py", line 507, in run
    await flasher.finish()
  File "/home/pi/katapult/scripts/flashtool.py", line 295, in finish
    await self.send_command("COMPLETE")
  File "/home/pi/katapult/scripts/flashtool.py", line 216, in send_command
    raise FlashError("Error sending command [%s] to Device" % (cmdname))
FlashError: Error sending command [COMPLETE] to Device
```

Lebensgefahr, 2025-02-28:
In my opinion, the utility should only do exactly what it is explicitly instructed to do. For example, it should not flash anything until a parameter (for example, -w) is provided, if it already has other parameters for interacting with the micro controller.

For example, -r requests the bootloader. It would be convenient to have a separate parameter for interaction, such as -q, which allows you to poll the bootloader (to confirm its availability or to check its parameters, such as the application offset). Only once you've confirmed that flashing is possible should you then provide the -w parameter.

Potentially dangerous operations should not occur automatically—for example, flashing some file from a directory that might not even be the firmware for this micro controller. Right now, that's exactly what happens. By default, it takes the file from ~/klipper/out/klipper.bin, and that file might very well not be the one intended for this micro controller, but something else. Therefore, I believe that the bin file should always be specified manually and not determined automatically.

EmperorArthur, 2025-02-28:
I agree with requiring at least a write argument, and that checks would be nice.  However, that is a different topic.  Especially since it would be a behavioral change which would break all existing script usage.

This one focuses on Katapult making the assumption that a CAN node running Klipper is also running Katapult via CAN.

Lebensgefahr, 2025-02-28:
> I agree with requiring at least a write argument, and that checks would be nice. However, that is a different topic. Especially since it would be a behavioral change which would break all existing [...]

Why don't you want to flash katapult with CAN mode too? Is it not possible?

EmperorArthur, 2025-03-03:
> Is it not possible?

No.

Klipper CAN adapter nodes are connected to the host by USB.  However, they appear as a CAN adapter, with the Klipper host as a CAN node.

Once you trigger bootloader mode, the CAN network goes down, and a new serial device shows up.