Arksine/katapult #153: Testing -r option and flash success?

cgarwood82, 2025-02-28:
I had to flash a new device today with katapult and I saw the new -r option so I wanted to give it a try with my AFC lite board. I built katapult and flashed using a typical method for an stm32 device. Then, I wanted to attempt to update klipper so I used the flashtool.py -r option with my device to get it into bootloader mode to do flash klipper. Instead it said there was a successful flash and the device name changed to now have katapult in the name instead of klipper. Was this expected? I was thinking it was going to kick it into bootloader only and I could then use flashtool to update klipper. You can see some of the confusion by the workflow in my unfiltered output here:

```mks@qidi-plus4:~/katapult/scripts$ sudo systemctl stop klipper
mks@qidi-plus4:~/katapult/scripts$ ./flashtool.py -d /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00  -r
Flashing Serial Device /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00, baud 250000
Detected USB device running Klipper
Requesting USB bootloader for /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00...
Waiting for USB Reconnect...Katapult detected
Flash Success
mks@qidi-plus4:~/katapult/scripts$ ./flashtool.py -d /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00  -f ~/klipper/out/klipper.bin 
Flashing Serial Device /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00, baud 250000
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/mks/katapult/scripts/./flashtool.py", line 875, in main
    await sock.run(args.device, args.baud, fpath, req_only)
  File "/home/mks/katapult/scripts/./flashtool.py", line 786, in run
    await self.validate_device(intf)
  File "/home/mks/katapult/scripts/./flashtool.py", line 695, in validate_device
    raise FlashError(f"No Serial Device found at {dev_path}")
FlashError: No Serial Device found at /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00
mks@qidi-plus4:~/katapult/scripts$ lsusb
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 006: ID 04d8:e72b Microchip Technology, Inc. Beacon RevH
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 003: ID 0735:0269 Asuscom Network 
Bus 004 Device 014: ID 1d50:6177 OpenMoko, Inc. stm32h723xx
Bus 004 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
mks@qidi-plus4:~/katapult/scripts$ ./flashtool.py -d /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00  -f ~/klipper/out/klipper.bin -r
Flashing Serial Device /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00, baud 250000
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/mks/katapult/scripts/./flashtool.py", line 875, in main
    await sock.run(args.device, args.baud, fpath, req_only)
  File "/home/mks/katapult/scripts/./flashtool.py", line 786, in run
    await self.validate_device(intf)
  File "/home/mks/katapult/scripts/./flashtool.py", line 695, in validate_device
    raise FlashError(f"No Serial Device found at {dev_path}")
FlashError: No Serial Device found at /dev/serial/by-id/usb-Klipper_stm32h723xx_37002C000551323532363237-if00
mks@qidi-plus4:~/katapult/scripts$ ls -l /dev/serial/by-
by-id/   by-path/ 
mks@qidi-plus4:~/katapult/scripts$ ls -l /dev/serial/by-id/usb-
usb-Beacon_Beacon_RevH_82A0EEA15157355957202020FF12201B-if00
usb-katapult_stm32h723xx_37002C000551323532363237-if00
mks@qidi-plus4:~/katapult/scripts$ ls -l /dev/serial/by-id/usb-^C
mks@qidi-plus4:~/katapult/scripts$ ./flashtool.py -d /dev/serial/by-id/usb-  -f ~/klipper/out/klipper.bin
usb-Beacon_Beacon_RevH_82A0EEA15157355957202020FF12201B-if00
usb-katapult_stm32h723xx_37002C000551323532363237-if00
mks@qidi-plus4:~/katapult/scripts$ ./flashtool.py -d /dev/serial/by-id/usb-katapult_stm32h723xx_37002C000551323532363237-if00  -f ~/klipper/out/klipper.bin
Flashing Serial Device /dev/serial/by-id/usb-katapult_stm32h723xx_37002C000551323532363237-if00, baud 250000
Detected USB device running Katapult
Detected Klipper binary version v0.12.0-450-g941fb5a36, MCU: stm32h723xx
Attempting to connect to bootloader
Katapult Connected
Software Version: v0.0.1-85-g5e11a9a
Protocol Version: 1.1.0
Block Size: 64 bytes
Application Start: 0x8020000
MCU type: stm32h723xx
Flashing '/home/mks/klipper/out/klipper.bin'...

[##################################################]

Write complete: 1 pages
Verifying (block count = 576)...

[##################################################]

Verification Complete: SHA = 2E1315ADC7AF64B05C4A98433F44781D1D3FAC3F
Flash Success
mks@qidi-plus4:~/katapult/scripts$ ls -l /dev/serial/by-id/usb-
usb-Beacon_Beacon_RevH_82A0EEA15157355957202020FF12201B-if00
usb-Klipper_stm32h723xx_37002C000551323532363237-if00
```

I got to where I needed to be, but the documentation doesn't really follow with that I experienced. Just wanted to make sure I'm using the tool right! Thanks for the great work. <3 katapult

Arksine, 2025-03-01:
Yes, the `Flash Success` message should be updated to account for the request type.  Katapult didn't flash anything, it just successfully executed the command to request the bootloader.  I will update the script when I get the opportunity.

cgarwood82, 2025-03-01:
Yeah, I saw that was the else statement at the end of  the script and figured that was the case. I was just a bit confused about the devise name change. Is that expected, just to inform the user that we are in fact in the bootloader?

I've been using a little script that does the following to enter bootloader mode and it's pretty convenient with no name changes: https://www.klipper3d.org/Bootloader_Entry.html#physical-serial

I'll run that before i run my katapult flashtool.py to hit my mcu's with new klipper and it's pretty effective. 

Coming back on topic, feel free to close this bug as I just wanted to check my understanding of the tool!

Arksine, 2025-03-01:
The `by-id` symbolic link will always change from Klipper's IDs to Katapult's when connected over virtual serial (USB).  If you want a stable symlink use the one in `by-path`.

The boot entry you linked above is for devices connected via the MCU's hardware UART, not virtual serial.   I have to presume you changed the way your board was connected, it shouldn't work for your current cofiguration.

cgarwood82, 2025-03-01:
Yeah, that's right. The above is for an AFC Lite board but the method I posted about earlier is what I am using for the machine's two MCU's and NOT the AFC Lite board I have in the output above. 

With that said, the -r option works for both virtual serial and hardware UART?

Arksine, 2025-03-02:
Yes, the `-r` option works for both.  Hardware UART's require it since Katapult cannot detect whether or not the device is running Klipper.   Tool goes through the following decision tree when `-r` is specified:

- Is the serial port a USB device?
  - no: use the [physical serial](https://www.klipper3d.org/Bootloader_Entry.html#physical-serial) bootloader request method
  - yes: Does the USB device have Klipper's USB IDs?
    - no: this is likely a USB-TTL adapter connected to the MCU's hardware UART. Use the `physical serial` bootloader request method
    - yes: use the [virtual serial](https://www.klipper3d.org/Bootloader_Entry.html#virtual-serial) bootloader request method
    
From the user perspective, you will see a change in USB IDs (and the `by-id` symbolic link) if the USB serial port is a "virtual" serial port configured by the MCU.  However, if the serial port is a USB-TTL adapter there will be  no change in USB IDs, thus no change in the symlink.

Lebensgefahr, 2025-03-03:
> Yes, the `-r` option works for both. Hardware UART's require it since Katapult cannot detect whether or not the device is running Klipper. Tool goes through the following decision tree when `-r` is [...]

It can be made more easy like here https://github.com/Arksine/katapult/pull/142 it is little bit tricky but it works and it is easier.
As changing baudrate and dtr is not distructive for physical serial it can be run first. You are a master, but this looks like unnecessary engineering.