Arksine/katapult #155: Unable to flash STM32H723 using CAN

sshantt, 2025-03-09:
I have been trying to use Katapult to update Klipper firmware on two CANbus Klipper boards I have. 

The boards have Katapult firmware installed. Katapult firmware is installed on there successfully. I can tell since the LED flashes when I rapid double click the reset button on either board. 

Klipper firmware is also installed correctly since it works just fine when I connect it to my raspberry pi with klipper software installed. 

I'm using a BTT U2C v2.1 as the CANbus adapter. I made sure to flash the latest firmware from here https://github.com/bigtreetech/U2C/tree/master/firmware

I made sure that both klipper and katapult make menuconfig are identical and correct. Here is what happens when I run the required katapult flashing code,

`
python3 flashtool.py -u d1669557c4ec -v
Connecting to CAN UUID d1669557c4ec on interface can0
Sending bootloader jump command...
Resetting all bootloader node IDs...
Detected Klipper binary version d7c0c933-dirty-20250308_230457-wb-pi-1, MCU: stm32h723xx
Attempting to connect to bootloader
INFO:root:Response for command CONNECT timed out, 4 tries remaining
INFO:root:Response for command CONNECT timed out, 3 tries remaining
INFO:root:Response for command CONNECT timed out, 2 tries remaining
INFO:root:Response for command CONNECT timed out, 1 tries remaining
INFO:root:Response for command CONNECT timed out, 0 tries remaining
INFO:root:Response for command COMPLETE timed out, 4 tries remaining
INFO:root:Response for command COMPLETE timed out, 3 tries remaining
INFO:root:Response for command COMPLETE timed out, 2 tries remaining
INFO:root:Response for command COMPLETE timed out, 1 tries remaining
INFO:root:Response for command COMPLETE timed out, 0 tries remaining
ERROR:root:Flash Tool Error
Traceback (most recent call last):
  File "/home/ubuntu/temp/katapult/scripts/flashtool.py", line 821, in run
    await flasher.connect_btl()
  File "/home/ubuntu/temp/katapult/scripts/flashtool.py", line 275, in connect_btl
    ret = await self.send_command('CONNECT')
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/temp/katapult/scripts/flashtool.py", line 403, in send_command
    raise FlashError("Error sending command [%s] to Device" % (cmdname))
FlashError: Error sending command [CONNECT] to Device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/temp/katapult/scripts/flashtool.py", line 1088, in main
    await sock.run()
  File "/home/ubuntu/temp/katapult/scripts/flashtool.py", line 831, in run
    await flasher.finish()
  File "/home/ubuntu/temp/katapult/scripts/flashtool.py", line 482, in finish
    await self.send_command("COMPLETE")
  File "/home/ubuntu/temp/katapult/scripts/flashtool.py", line 403, in send_command
    raise FlashError("Error sending command [%s] to Device" % (cmdname))
FlashError: Error sending command [COMPLETE] to Device
`

Here is what the candump looks like for that same time period

`
(1741482567.763425) can0 3F0#02D1669557C4EC
(1741482568.764713) can0 3F0#12
(1741482569.337230) can0 3F0#11D1669557C4EC81
(1741482569.838014) can0 202#01881100F17C9903
(1741482572.591518) can0 202#01881100F17C9903
(1741482575.344850) can0 202#01881100F17C9903
(1741482578.098189) can0 202#01881100F17C9903
(1741482580.851475) can0 202#01881100F17C9903
(1741482583.604776) can0 202#01881500911B9903
(1741482586.358102) can0 202#01881500911B9903
(1741482589.111407) can0 202#01881500911B9903
(1741482591.864671) can0 202#01881500911B9903
(1741482594.618067) can0 202#01881500911B9903
`

Here is the can setup

`
ip -details link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0  allmulti 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 0 
	 bitrate 1000000 sample-point 0.750
	 tq 62 prop-seg 5 phase-seg1 6 phase-seg2 4 sjw 2 brp 4
	 gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp_inc 1
	 termination 0 [ 0, 120 ]
	 clock 64000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 parentbus usb parentdev 3-2:1.0
`

Here is what shows up when I query with the flashtool.py

`
python3 flashtool.py -q        
Resetting all bootloader node IDs...
Checking for Katapult nodes...
Detected UUID: 48ca7afe7a44, Application: Klipper
Detected UUID: d1669557c4ec, Application: Klipper
CANBus UUID Query Complete
`

One thing to note is that as soon as I run the command, the board led starts flashing showing that katapult flash mode is active. So it seems like katapult is able to get the MCU into katapult flash mode but is not able to carry on from then on out for some reason. 

I tried two new boards, with a new usb to can adapter and a new git clone of both repos and a new make compile firmware with the same results. I also tried flashing from a different linux machine.

I have included screenshots of the menu makeconfig of both for your reference.

[image]
[image]

Am I missing something? I feel like I'm either doing something wrong or there's something in the katapult code that is preventing katapult from connecting to that particular MCU on those particular CAN pins

thunderkeys, 2025-06-01:
We've been seeing this a fair bit on the ArmoredTurtle discord of late, for flashing Klipper to the AFC-Lite (STM32H723 board) after successfully flashing Katapult.

Reflashing an older Katapult commit version (e.g. e1716657) and using the same Klipper-compiled firmware binary has worked.

I suspect (but have not yet had time to verify) that it's the changes to flashtool.py in 28b873de that may be causing this (which was committed 3 days before this issue was opened).

Arksine, 2025-07-22:
@fer662 @thunderkeys I was able to reproduce this "send block" issue on a BTT Kraken (H723 MCU).  It doesn't appear to be an issue with Katapult or the flashtool, instead there seems to be some errata on the MCU itself with regard to in application programming.  I was able to resolve it by performing a full chip erase prior to flashing Katapult.  I used stm32 cube programmer to do this via dfu, I'm not sure if using dfu-util will work but its worth a try if that is your tool of choice.

fer662, 2025-07-22:
Does that explain why the old version of Katapult works though?

Arksine, 2025-07-22:
> Does that explain why the old version of Katapult works though?

 I wasn't able to get any prior version of Katapult to work prior to the mass erase.

fer662, 2025-07-22:
Interesting. Maybe I did the mass erase as a means to get back into a state to flash the older Katapult and that solved it? I cannot remember

AlatusAles, 2025-07-23:
Я поддержу версию о полном стирании чипа, только с этим аргументом удалось прошить фабричный октопус с дфю-утилл

Arksine, 2025-07-23:
> Interesting. Maybe I did the mass erase as a means to get back into a state to flash the older Katapult and that solved it? I cannot remember

Its possible.  I reviewed all of the changes to the flashtool looking for anything I may have missed.  I did reduce the default command timeout in commit a8de2feab739080388b575bf6dce01c364ecd830 from 5 seconds to 2 seconds.

Two seconds should be plenty of time for a response, but I did notice that the stm32h723 seems to take longer than expected to write the first block.  Its possible that in some circumstances the command will time out, and the next try enables write protection on that page.  In commit 399e50e1210062a297dfb93c836731de150cbdc6 I set the timeout for the `send block` command back to 5 seconds.

fer662, 2025-07-23:
Sadly i don't have a board to help you debug. This was done over ssh on someone else's printer.

Arksine, 2025-07-23:
@AlatusAles I apologize, but you will need to add your feedback in English.

FWIW, the `send block` error seems unrelated to the OPs issue, which appears to be a general problem with communication.  The candump shows no response from the MCU, so I'm not quite sure what is happening.  I would recommend pulling the latest version of Katapult.

With regard to the `send block` errors, I believe they are more relevant in issue #128, so I think it would be beneficial to move further discussion pertaining to that particular issue there.