Arksine/katapult #160: Issue flashing MCU of SV08, Hangs at Detected USB device running Katapult

makerwolf, 2025-04-04:
I am trying to flash the klipper via katapult on my SV08 and I am having big trouble. I've done the exact same process following this guide, and the toolhead flashed just fine, as expected. But the main MCU just hangs at "Detected USB device running Katapult". 
I've spent 3 hours now researching and trying different approaches but nothign seems to work. There is no webinterface open, I have also unplugged the default screen on EXP1 and EXP2. It just wont go past there and I cannot find anything wrong. 

Any ideas what I could do?

thiagoshima, 2025-05-18:
Same, after a few hours stuck I was able to flash successfully using

`python3 flash_can.py -d /dev/serial/by-id/usb-katapult_stm32f103xe_50FF67068367515532431367-if00 -b 250000
`
 the script is in the katapult folder

thiagoshima, 2025-06-09:
> > Same, after a few hours stuck I was able to flash successfully using
> > `python3 flash_can.py -d /dev/serial/by-id/usb-katapult_stm32f103xe_50FF67068367515532431367-if00 -b 250000 ` the script is in the katapult folder
> 
> Hey there! I had this problem too, but when I ran the command it looks like klipper.bin is not being made as the path name is wrong and when I go to klipper/out klipper.bin is not there. Any idea how to move forward?

make sure that you got katapult flashed before trying the command, also you need to use the serial-id of your board, check the klipper docs on how to find it

McMuppet, 2025-06-10:
Now when I list my usb devices I see one listed with 'katapult' and the other with 'Klipper' in the names. I have updated the toolhead just fine, so is this part of the problem having two different names or is that normal if the toolhead is flashed but not the mainboard? 

[image]