Arksine/katapult #166: make: *** No rule to make target 'out/src/generic/timer_irq.o', needed by 'out/katapult.elf'.  Stop.

Bones116XIVI, 2025-05-19:
im going through the steps to flash my EBB36 and i keep getting this message after i enter the make cmd. im in the correct directory. i entered the correct data in the menuconfig, i tried to make clean and re compile and still to no avail. i am not that proficient as far as trouble shooting these issues, is there anyone that could help me out here?

pgsengstock, 2025-05-20:
Seeing the same thing trying to build for an M8P.  Running origin/master, freshly cloned.  Combing through the recent commits now, but not spotting anything obvious.

pgsengstock, 2025-05-20:
Narrowed down to commit 3f84643.

pgsengstock, 2025-05-20:
Not sure, but I think src/stm32/Makefile need lines 42-43 edited as follows:
`timer-src-$(CONFIG_MACH_STM32F0) := stm32/stm32f0_timer.c
timer-src-$(CONFIG_MACH_STM32G0) := stm32/stm32f0_timer.c`

At least, that makes it build properly.  No clue if it breaks anything.

KirkSturg, 2025-05-20:
I too am having the same issue.  Based on the error it looks likes some kind of file issue or spelling issue.  Anyone have a update for a fix?

spamwax, 2025-05-20:
Getting the same error (compiling for an EBB42). it seems the organizational change made in that commit has broken the Makefile

Arksine, 2025-05-20:
Thanks.  This should be fixed in commit a8e44fd3089cf94cb0ace970c4148f6aab078cc0.

Arksine, 2025-05-20:
Thanks.  The deployer build errors should now be fixed as well.

Bones116XIVI, 2025-05-21:
still having issues with the makefile  Linking out/deployer.elf
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: out/src/generic/armcm_boot.o: in function `reset_handler_stage_two':
/home/bones/katapult/src/generic/armcm_boot.c:103: undefined reference to `armcm_main'
collect2: error: ld returned 1 exit status
make: *** [Makefile:99: out/deployer.elf] Error 1
this is definitely putting a damper on my printers pushing plastic capabilities lol.

Printathan, 2025-05-21:
Worked for me just now after updating

Bones116XIVI, 2025-05-21:
what steps did you take

Bones116XIVI, 2025-05-21:
im not the most proficient

Bones116XIVI, 2025-05-21:
sudo apt update and upgrade?

Printathan, 2025-05-21:
I'm following Esoterical's can bus guide and just started back at "Installing Katapult" and ran `test -e ~/katapult && (cd ~/katapult && git pull) || (cd ~ && git clone https://github.com/Arksine/katapult) ; cd ~
`

Bones116XIVI, 2025-05-21:
how do i remove the katapult directory i tried sudo rm ~/katapult

Bones116XIVI, 2025-05-21:
yeah im still getting the same error

pgsengstock, 2025-05-21:
Worked for me just fine after updating. From inside the cloned repo, just run `git pull` to update.

spamwax, 2025-05-21:
> yeah im still getting the same error

a clean up should fix it, this will work if `katapult` was originally in your home folder:

```bash
cd ~
rm -rf katapult
git clone https://github.com/Arksine/katapult
cd katapult
make menuconfig # adjust the setting according to your board
make clean
make
```

Bones116XIVI, 2025-05-21:
thanks man

infinitevalence, 2025-05-21:
Updated my local repo, and this fixed the issue.  Thanks!